morse-simulator/morse #591: Shadows are a little off

slremy, 2015-01-20:
Hello! I've finally gotten around to testing morse and things look really nice! One quick question, is there something that I did wrong that would have resulted in the shadows being a little off in quadrotor\_dynamic\_example.py? For the included image, I added a camera via:

     camera = VideoCamera()
     camera.translate(x=0.2, z=0.9)

[image]

adegroote, 2015-02-02:
Do you see a real difference between the viewport camera and the rendered camera ? Shadow looks relatively ok, considering the position of the light (Light2 in the named scenario). 

If you see a real difference between viewport and rendered camera, it is probably a Morse bug, but I don't really see that. If you think the shadow are not good considering light position, it is a more general blender issue

adegroote, 2015-04-07:
Without more precision on your issue, we cannot fix it, and I will close the issue. Thanks in advance.

slremy, 2015-04-07:
I'd gotten swamped and missed the earlier update. Sorry!

severin-lemaignan, 2015-04-07:
@slremy No worries :-) Can you answer @adegroote question?

slremy, 2015-04-07:
I will re-image a machine and test this morning.
Should be able to report back with an update this afternoon.

slremy, 2015-04-07:
The configuration (or something) has changed since I initially did these
tests.
On a fresh install I now get:

  File "/usr/local/lib/python3/dist-packages/morse/builder/bpymorse.py",
line 315, in set_speed
    get_context_scene().game_settings.fps = fps
AttributeError: 'NoneType' object has no attribute 'game_settings'

I will have to agree with Arnaud's earlier suggestion and close my original
issue since I won't be able to track down this new issue today.

Thank you for your attention.

adegroote, 2015-04-07:
On 07/Apr - 11:05, slremy wrote:
> The configuration (or something) has changed since I initially did these
> tests.
> On a fresh install I now get:
>
>   File "/usr/local/lib/python3/dist-packages/morse/builder/bpymorse.py",
> line 315, in [...]

Can you paste your builder script ?

> 
> I will have to agree with Arnaud's earlier suggestion and close my original
> issue since I won't be able to track down this new issue today.

It is not a big issue if you can't test it today, it is just a reminder
to avoid having database saturated by bug with no-information and
no-answer from the original submitter.

slremy, 2015-04-07:
On Tue, Apr 7, 2015 at 2:22 PM, Arnaud Degroote <notifications@github.com>
wrote:

> On 07/Apr - 11:05, slremy wrote:
>
> Can you paste your builder script ?

This is the script that I used to install ROS and morse:
https://remy-robotics-tools.googlecode.com/svn/trunk/ros_morse/install.sh
The object that I tested (then and now) was a quadrotor:
https://remy-robotics-tools.googlecode.com/svn/trunk/ros_morse/ros_quadrotor.py

adegroote, 2015-04-08:
On 07/Apr - 12:12, slremy wrote:
> On Tue, Apr 7, 2015 at 2:22 PM, Arnaud Degroote <notifications@github.com>
> wrote:
>
>
> This is the script that I used to install ROS and morse: [...]
The scene seems to work properly on my Morse installation

adegroote, 2015-04-14:
Considering the discussion on morse-users@, can I close it ?

slremy, 2015-04-14:
Yes.
Thank you.

closed 2015-04-14 by adegroote