morse-simulator/morse #592: Build is not reproducible

sylvestre, 2015-01-31:
Debian has an effort to make every packages reproducible in term of binary.
Error:
```
 Python doctrees writes a `environment.pickle.gz`, in a random order.
```
See:
https://reproducible.debian.net/rb-pkg/morse-simulator.html

adegroote, 2015-02-02:
Can you be a bit more precise about the issue ? I don't have this specific file in any of my Morse build (through I have some .doctrees/environment.pickle). 

Does debian has some kind of fix for this issue ? It is not something really specific to Morse, but to all projects using sphinx to build their documentation

adegroote, 2015-04-07:
Sphinx issue, as noted in https://wiki.debian.org/ReproducibleBuilds/SphinxIssues. So definitively not a Morse bug. I close it.

closed 2015-04-07 by adegroote