morse-simulator/morse #593: Stereo Camera not launching

tiberiusferreira, 2015-02-04:
Hello, thanks for building Morse and this awesome community! I'm having a problem with the stereo camera setup (or any camera in general, even just one).
 When I try to use this code which is from the tutorials:
 		# A pan-tilt unit to be able to orient the cameras
		Platine = PTU()
		Platine.translate(x=0.2000, z=0.9000)
		self.append(Platine)

		# The STEREO UNIT, where the two cameras will be fixed
		#Stereo = StereoUnit()
		#Stereo.translate(z=0.0400)
		#Platine.append(Stereo)
		
		# Left camera
		CameraL = VideoCamera()
		CameraL.translate(x=0.1000, y=0.2000, z=0.0700)
		self.append(CameraL)
		CameraL.properties(capturing = True)
		CameraL.properties(cam_width = 320)
		CameraL.properties(cam_height = 240)
		CameraL.properties(cam_focal = 25.0000)
		CameraL.add_stream("ros",topic='cameraL')

		# Right camera
		CameraR = VideoCamera()
		CameraR.translate(x=0.1000, y=-0.2000, z=0.0700)
		self.append(CameraR)
		CameraR.properties(capturing = True)
		CameraR.properties(cam_width = 320)
		CameraR.properties(cam_height = 240)
		CameraR.properties(cam_focal = 25.0000)
		CameraR.add_stream("ros",topic='cameraR')

It loads up the simulation, then freezes it and doesn't start, but the terminal log says it started normally. Then it displays terminal text as if it was reloading again, then it starts but with errors of unset variables as if my models did not go through the setup phase. 
 My work friends experienced the same problem and they say it is fixed replacing the camera scripts by older versions: 1.0 or 1.0.1 . It appears to be that the cameras create additional scenes in morse and in those scenes my models don't go through the regular setup phase. It also causes very poor performance.
[image]

adegroote, 2015-02-04:
Can you post the complete builder script, the error does not seems to be related directly to the camera, but with some interaction between robot definition and the camera ?

tiberiusferreira, 2015-02-04:
Sure. 
import sys
sys.path.append("../teste8")

from Carina2Class import *
from sensor import *
from morse.builder import *

class Carina2Builder(Robot):
	def __init__(self):
		Robot.__init__(self, "carina2.blend")		
		self.properties(classpath="Carina2Class.Carina2Class")
		
		#add odometry
		odometry = Odometry()
		#odometry.translate(0,0,0)
		self.append(odometry)
		odometry.add_stream('ros', topic='/odom')

		
		pose = Pose()
		pose.translate(0,1.3,0.2)
		self.append(pose)		
		#pose.add_stream("ros", method="morse.middleware.ros.pose.TFPublisher", topic="/base_link")
		pose.add_stream("ros", topic="/pose")
		
		# add sick sensor
		sick = Sick()
		sick.translate(0,2.6,0.2)
		sick.rotate(0,0,1.570796327)
		self.append(sick)
		sick.add_stream("ros", topic='/sick')
		
		#add gps
		gps = GPS()
		gps.translate(0,1.3,0.2)
		self.append(gps)
		gps.add_stream("ros", topic='/gps')
		
		#add imu
		imu = IMU()
		imu.translate(0,1.3,0.2)
		self.append(imu)
		imu.add_stream("ros", topic='/imu')
		
		#velocity
		velocity = Velocity()
		velocity.translate(0,1.3,0.2)
		self.append(velocity)
		velocity.add_stream("ros", topic='/velocity')		
		
		

		# A pan-tilt unit to be able to orient the cameras
		Platine = PTU()
		Platine.translate(x=0.2000, z=0.9000)
		self.append(Platine)

		# The STEREO UNIT, where the two cameras will be fixed
		Stereo = StereoUnit()
		Stereo.translate(z=0.0400)
		Platine.append(Stereo)
		
		# Left camera
		CameraL = VideoCamera()
		CameraL.translate(x=0.1000, y=0.2000, z=0.0700)
		Stereo.append(CameraL)
		CameraL.properties(capturing = True)
		CameraL.properties(cam_width = 320)
		CameraL.properties(cam_height = 240)
		CameraL.properties(cam_focal = 25.0000)
		CameraL.add_stream("ros",topic='cameraL')

		# Right camera
		CameraR = VideoCamera()
		CameraR.translate(x=0.1000, y=-0.2000, z=0.0700)
		Stereo.append(CameraR)
		CameraR.properties(capturing = True)
		CameraR.properties(cam_width = 320)
		CameraR.properties(cam_height = 240)
		CameraR.properties(cam_focal = 25.0000)
		CameraR.add_stream("ros",topic='cameraR')

			
		

		print("bge alive")

adegroote, 2015-02-04:
Can you provide also carina2.blend and the python file defining Carina2Class ? It seems the issue is somewhere between these two files, because it works perfectly when I use morsy.blend and Morsy class instead of them.

tiberiusferreira, 2015-02-05:
Sure. Here is the Carina2Class : https://www.dropbox.com/s/kh2jc4xgw9sfq83/Carina2Class.py?dl=0

 Here is the carina2.blend https://www.dropbox.com/s/69oh03w69fdgh5u/carina2.blend?dl=0 

 Thanks for the help!

adegroote, 2015-02-05:
After investigating a bit more, the problem comes from the interaction from your scripts and the fact that camera creates several scenes. 

But in these new scenes, physics is desactived, (but not logic, probably because it is not possible). These scenes are necessary to properly render scene with the camera resolution, otherwise the camera image is based from the viewport rendering and then scaled to match the camera resolution.

In Morse, the logic is not executed in other scene, due to the logic added in calling

https://github.com/morse-simulator/morse/blob/master/src/morse/blender/calling.py#L33

In your case, logic is executed in the different scenes (which probably is the cause of poor performance). But as the physics is deactived in other scene, physics API fails more or less properly. By adding the same logic than in calling.py, you should fix your issue (basically get scene_name, and only call main() if scene_name is S.MORSE_LOGIC).  Of course, stuff would be easier if you use the "actuator" logic of Morse (it looks like  your code is an extension/improvment over "wheeled_robot" / "steer_force" actuator).

tiberiusferreira, 2015-02-05:
Thanks for the help! 
 I tried fixing in the logic actuator, I put 

 def main():
    # if current scene is not the main MORSE logic scene, pass
	if bge.logic.getCurrentScene().name != 'S.MORSE_LOGIC':
		print('AllSET EQUALS 0!\n\n\n\n\n\n\n')    
		Allset=0
		return
	
	
	Allset=1
	# get car object
	print('AllSET EQUALS 1!\n\n\n\n\n\n\n')

In my main in the CarSetup.py script inside carina2.blend
Then if it is set to 1 I trigger the state switch to the regular logic, but it never gets is set to 1. Am I doing something wrong? 

 Here is the carina2.blend https://www.dropbox.com/s/69oh03w69fdgh5u/carina2.blend?dl=0

adegroote, 2015-02-06:
You need to add the 

'''
if bge.logic.getCurrentScene().name != 'S.MORSE_LOGIC':
        return
'''
logic in all main function, not only the setupCar one.

tiberiusferreira, 2015-02-09:
Thank you VERY much!!!! It worked!!

closed 2015-02-09 by tiberiusferreira