morse-simulator/morse #595: morse create actuator/sensor seems to be broken

nicolaje, 2015-02-11:
The actuator/sensor creation template tool seems to be broken since quite a few commits.

Using:
* the latest morse development version: morse 1.2-173-g1795b
* python 3.4 Python 3.4.0
* Blender v.2.69.0
* Ubuntu 14.04

Issuing:
* morse create test
* morse add sensor sTest test
* with the following builder script:
```
from morse.builder import *
from test.builder.sensors import Stest

sTest=Stest()

robot = Morsy()

env = Environment('sandbox', fastmode = False)
```
* morse run test fails this way:
```
[ERROR][MORSE] Uncaught exception, quit Blender.
Traceback (most recent call last):
  File "/home/jeremy_nicola/morse/build/test/default.py", line 6, in <module>
    sTest=Stest()
  File "/home/jeremy_nicola/morse/build/test/src/test/builder/sensors/sTest.py", line 7, in __init__
    "sTest")
  File "/usr/local/lib/python3/dist-packages/morse/builder/creator.py", line 119, in __init__
    make_morseable = make_morseable)
  File "/usr/local/lib/python3/dist-packages/morse/builder/creator.py", line 66, in __init__
    obj.game.physics_type = 'NO_COLLISION'
AttributeError: 'NoneType' object has no attribute 'game'
Traceback (most recent call last):
  File "/home/jeremy_nicola/morse/build/test/default.py", line 6, in <module>
    sTest=Stest()
  File "/home/jeremy_nicola/morse/build/test/src/test/builder/sensors/sTest.py", line 7, in __init__
    "sTest")
  File "/usr/local/lib/python3/dist-packages/morse/builder/creator.py", line 119, in __init__
    make_morseable = make_morseable)
  File "/usr/local/lib/python3/dist-packages/morse/builder/creator.py", line 66, in __init__
    obj.game.physics_type = 'NO_COLLISION'
AttributeError: 'NoneType' object has no attribute 'game'
```

The same happens with actuators.

This behaviour was introduced since b2708d5641b7cf82fd600fd4794001fedfb22212

adegroote, 2015-02-11:
Can you confirm it fixes your issue ?  Thanks.

nicolaje, 2015-02-12:
It fixes the issue, thanks!

commit 5e7c2092 (2015-02-12): [templates] Handle properly b2708d5641 in sensor/actuator template

Fix #595
closed 2015-02-12 by adegroote