morse-simulator/morse #598: Update install doc

pierriko, 2015-02-23:
To mention that ROS supports requires source, to avoid monthly mail like: 
https://sympa.laas.fr/sympa/arc/morse-users/2015-02/msg00011.html

adegroote, 2015-02-25:
Is the sentence from installation not sufficient ? 

```
If you plan to use the simulator with raw sockets or text files as interface (for instance, to integrate MORSE with MatLab or other specific applications), you don’t need anything else, and you can jump to MORSE’s Quickstart. Otherwise, you need to install the software for the desired middlewares
```

pierriko, 2015-02-25:
regarding the number of ros-support related emails, I'd say no ;-)

pierriko, 2015-02-25:
hopefully https://github.com/morse-simulator/morse/commit/8f330a75099cf37227ea4b1ee4b9ab2c15e83c00 will help

commit 11ee3458 (2015-04-07): [doc] Rewritten (and simplified) installation notes

Should fix #598
closed 2015-04-07 by severin-lemaignan