morse-simulator/morse #599: Nogravity and groundrobot properties needed for correct physics?

eirikhex, 2015-02-24:
I am trying to implement som additional physics to a underwater robot an experience the following problems: 

If I do not use a GroundRobot (or set the ground robot property) the forces do not create an acceleration, just a constant velocity as long as the force is acting (not following newtons 2. Law). It is also moving relatively slow.

If the NoGravity property is set to True I experience the same problem, (even if Groundrobot is True)

With NoGravity set to True the robot is still falling slowly, even if I have not applied any other forces than setting the gravity constant. (same as #551 )

With the mass of the robot set to M and the gravity set to g in the builder environment, and an added buoyancy force of M*g (positive) the robot is not standing still, even in theory these forces should counteract eachother 

All forces are applied through bge_object.applyForce()

What do the groundRobot and Nogravity properties actually do, and are these properties processed by morse or by blender? Is what I am experiencing a bug, or is it simply that the property names groundrobot and nogravity is misleading, and actually need to be enabled to get proper evaluation of the applied forces?

adegroote, 2015-02-24:
GroundRobot and NoGravity are properties which are passed almost as it to morse.core.robot.Robot, and named in the python code _is_ground_robot and _no_gravity. These properties are only used in the ``apply_speed`` method (https://github.com/morse-simulator/morse/blob/master/src/morse/core/robot.py#L50) which is used in several actuators (joystick, keyboard, waypoint, destination, xy_omega, v_omega). Through, if you experiment issues with your own actuator, it probably means you have multiple actuators on the same 'robot', which some mixed-control in force and in speed, and I'm not sure to understand how it is handled in Blender/Bullet (or if it supposed to work at all).

 You should probably use only 'force-based' control (such as rotorcraft_... and steer_force).  For these actuators, GroundRobot and NoGravity has no impact.

I hope it helps. Otherwise, feel free to ask again, will try to be clearer.

eirikhex, 2015-02-24:
Thank you the prompt answer. 

You are right. It was the v_omega actuator that was added in the builder template that was causing the problems. Now it works like a charm (exept the last issue, but that is minor, and blender specific anyway)

closed 2015-02-24 by eirikhex

adegroote, 2015-02-24:
On 24/Feb - 10:34, Eirik Hexeberg Henriksen wrote:
> Thank you the prompt answer. 
> 
> You are right. It was the v_omega actuator that was added in the builder template that was causing the problems. Now it works like a charm (exept the last issue, but that is minor, and blender specific anyway)
> 
Which issue exactly ?

-- 
Arnaud Degroote
ISAE DISC

eirikhex, 2015-02-24:
This issue:
With the mass of the robot set to M and the gravity set to g in the builder environment, and an added buoyancy force of M*g (positive) the robot is not standing still, even in theory these forces should counteract eachother

The robot is experiencing a small downward force. I assume this is blender specific because when I read the blender internal mass and gravity variables they are not exactly equal to the ones that are set. (Gravity is typically 9.8100000362..)

However even if I set the bouyancy force equal to blender mass times blender gravity i get a small movement. I assume that it is an internal blender issue.. And it is no problem for my simulation as the difference is very small.