morse-simulator/morse #600: Failing to load modules from ros_request_manager.py

hawesie, 2015-03-01:
This is on OS X 10.10.2 using a source install of [our fork of](https://github.com/strands-project/morse) MORSE from  `morse check` reports

```
* Checking up your environment...

* MORSE has only been tested on Linux. It may work
on other operating systems as well, but without any guarantee

* Found MORSE libraries in '/opt/morse/lib/python3.4/site-packages/morse/blender'. Alright.

* Trying to figure out a prefix from the script location...

* Default scene found. The prefix seems ok. Using it.

* Setting $MORSE_ROOT environment variable to default prefix [/opt/morse/]

* Checking version of /Applications/Blender/blender.app/Contents/MacOS/blender... Found v.2.71.0

* Blender found from $MORSE_BLENDER. Using it (Blender v.2.71.0)

* Checking version of Python within Blender /Applications/Blender/blender.app/Contents/MacOS/blender... Found v.3.4.0

* Blender is compiled for Python 3.4.0 but MORSE has been compiled for Python 3.4.3! Check your MORSE build configuration or the selected Blender version.

* Your environment is correctly setup to run MORSE.
```

My python versions mismatch due to #576 -- but MORSE seems to start up fine in non-ROS cases. When I try to run the ROS tutorial I get errors importing modules from `ros_request_manager.py` (`roslib`, `rospy`, lines 12, 13). Line 12 errors with `No module named expat; use SimpleXMLTreeBuilder instead`, line 13 with `No module named 'xmlrpclib'`. The problem seems to be related to mismatches in versions of Python 2.* to 3.* but I'm having difficulty tracking down the problem beyond this. Does anyone have any clues or things for me to check?

adegroote, 2015-03-02:
Does ros_stream_manager works ?
Can you print your $PYTHONPATH ? 
Do you follow the instruction from https://www.openrobots.org/morse/doc/stable/user/installation/mw/ros.html ?

hawesie, 2015-03-02:
> Does ros_stream_manager works ?

How do I test that? The only reason I know `ros_request_manager.py` is where the problem starts is that was printing stuff I could grep for.

> Can you print your $PYTHONPATH ? 

In the terminal it is

```
/opt/ros/indigo/lib/python2.7/site-packages
```

If I print `sys.path` from ros_request_manager.py I get

```python
['/opt/morse/share/morse/data', '/opt/morse/share/morse/data/props', '/opt/morse/share/morse/data/environments/indoors-1', '/opt/morse/share/morse/data/robots', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/addons', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/startup', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/modules', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/morse/bin', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/pip-6.0.8-py3.4.egg', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/ros/indigo/lib/python2.7/site-packages', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python34.zip', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/plat-darwin', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/lib-dynload', '/usr/local/lib/python3.4/site-packages', '/opt/morse/lib/python3.4/site-packages', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python34.zip', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/plat-darwin', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/lib-dynload', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/site-packages', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python34.zip', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/freestyle/modules', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/addons/modules', '/Users/nah/Library/Application Support/Blender/2.71/scripts/addons/modules', '/opt/morse/lib/python3.4/site-packages/morse/blender', '/opt/morse/lib/python3.4/site-packages/morse/blender/human_interaction']
```

Do you follow the instruction from https://www.openrobots.org/morse/doc/stable/user/installation/mw/ros.html ?

I installed rospkg, catkin etc. using pip3. They are installed in `/usr/local/lib/python3.4/site-packages` and are being found by MORSE (if not I see a message about missing rospkg).

adegroote, 2015-03-02:
On 02/Mar - 01:23, Nick Hawes wrote:
> > Does ros_stream_manager works ?
> 
> How do I test that? The only reason I know `ros_request_manager.py` is where the problem starts is that was printing stuff I could grep for.

Does a scene with only a add_stream('ros') entry work (in comparison to
a scene with add_service('ros') (or add_interface('ros').
> In the terminal it is
>
> ```
> /opt/ros/indigo/lib/python2.7/site-packages
> ```
>
> If I print `sys.path` from ros_request_manager.py I get
>
> ```python
> ['/opt/morse/share/morse/data', [...]

This path does not appear in PYTHONPATH not sys.path, so I think it is
probably the cause of the issue. Morse is probably trying to import
rospkg from /opt/ros/indigo/lib/python2.7/site-packages which probably
lead to these errors.

hawesie, 2015-03-02:
I made this edit to the tutorial file:

```python
# Add default interface for our robot's components
# atrv.add_default_interface('ros')
atrv.add_stream('ros')
```

The simulation starts fine, but there are no ROS topics.

> This path does not appear in PYTHONPATH not sys.path, so I think it is
probably the cause of the issue. Morse is probably trying to import
rospkg from /opt/ros/indigo/lib/python2.7/site-packages which probably
lead to these errors.

`/usr/local/lib/python3.4/site-packages` is the default package path for python3 on my system, so I think it's found anyway. Either way adding it explicitly doesn't change the behaviour.

adegroote, 2015-03-02:
``atrv.add_stream('ros')`` does not do anything useful. You should put on a sensor.

I missed ``/usr/local/lib/python3.4/site-packages`` in your sys.path, but in any case, it is after ``/opt/ros/indigo/lib/python2.7/site-packages`` so python takes roslib and cie from /opt/ros/indigo... (and so the 2.7 stuff)

hawesie, 2015-03-02:
Adding the stream to pose:

```
[SCENE INITIALIZATION]
[    0.092] Component: 'atrv.pose' using datastream 'morse.middleware.ros_datastream.ROSDatastreamManager'
[    0.093] 	Datastream interface 'morse.middleware.ros_datastream.ROSDatastreamManager' created
[helpers.loading] Module not found: No module named 'xmlrpclib'
[helpers.loading] Could not load the class PoseStampedPublisher in morse.middleware.ros.pose
[helpers.loading] Could not create an instance of morse.middleware.ros.pose.PoseStampedPublisher
```

This looks like the same error: `[helpers.loading] Module not found: No module named 'xmlrpclib'`

The python path I use when launching this is: `/usr/local/lib/python3.4/site-packages:/opt/ros/indigo/lib/python2.7/site-packages`. If I don't have the indigo 2.7 path then `rospy` is not found. 

The python path I see at runtime is:

```
[PRE-INITIALIZATION]
[    0.014] Python Version: 3.4.0
[    0.014] Blender Version: 2.71.0
[    0.014] Python path: ['/opt/morse/share/morse/data', '/opt/morse/share/morse/data/props', '/opt/morse/share/morse/data/environments/indoors-1', '/opt/morse/share/morse/data/robots', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/addons', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/startup', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/modules', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/morse/bin', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/pip-6.0.8-py3.4.egg', '/usr/local/lib/python3.4/site-packages', '/opt/ros/indigo/lib/python2.7/site-packages', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python34.zip', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/plat-darwin', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/lib-dynload', '/opt/morse/lib/python3.4/site-packages', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python34.zip', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/plat-darwin', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/lib-dynload', '/Applications/Blender/blender.app/Contents/MacOS/2.71/python/lib/python3.4/site-packages', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/opt/ros/indigo/lib/python2.7/site-packages/python_qt_binding-0.2.14-py2.7.egg', '/usr/local/lib/python3.4/site-packages/setuptools-12.2-py3.4.egg', '/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python34.zip', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/freestyle/modules', '/Applications/Blender/blender.app/Contents/MacOS/2.71/scripts/addons/modules', '/Users/nah/Library/Application Support/Blender/2.71/scripts/addons/modules', '/opt/morse/lib/python3.4/site-packages/morse/blender', '/opt/morse/lib/python3.4/site-packages/morse/blender/human_interaction']
```

adegroote, 2015-03-03:
Can you check the content of your ros_comm/core.py ? Does it contains

```
try:
import xmlrpc.client as xmlrpcclient #Python 3.x
except ImportError:
import xmlrpclib as xmlrpcclient #Python 2.x
```

Does 

```import xmlrpc.client``` works in the Morse context ?

hawesie, 2015-03-03:
> On 3 Mar 2015, at 08:48, Arnaud Degroote <notifications@github.com> wrote:
> 
> Can you check the content of your ros_comm/core.py ? Does it contains
> 
> try:
> import xmlrpc.client as xmlrpcclient #Python 3.x
> except ImportError:
> import xmlrpclib as xmlrpcclient #Python 2.x
> 
> 

Yes, that’s in there.

> import xmlrpc.client works in the Morse context ?
> 

Interesting! Adding that to ros_request_manager.py generates a new error. I will pursue this a bit and see what I can find wrong.

```
[helpers.loading] Module not found: dlopen(/usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/lib-dynload/pyexpat.so, 2): Symbol not found: __PyTraceback_Add
  Referenced from: /usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/lib-dynload/pyexpat.so
  Expected in: flat namespace
 in /usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/lib-dynload/pyexpat.so
```

hawesie, 2015-03-03:
No change on a clean install. Do you think it could be because of the python versions mismatch 3.4.0 (blender) vs 3.4.3?

adegroote, 2015-03-03:
Yes, it is a new "API" introduced in 3.4.3 to fix http://bugs.python.org/issue22462 (commit https://hg.python.org/cpython/rev/f2f13aeb590a).

Another proof we definitively need to check minor version of Python

hawesie, 2015-03-03:
Resolved! As discussed in https://github.com/morse-simulator/morse/issues/576 when the 3.4.0 versions match there is the error `Fatal Python error: PyThreadState_Get: no current thread`. 3.4.3 gives the above problem. I am now running the  bottled python3 install 3.4.2_1 from Homebrew alongside Blender 2.71.0 compiled with Python 3.4.0. This appears to be working. 

Thanks for you help @adegroote

adegroote, 2015-03-03:
>  This appears to be working. 

Until the next python mismatch issue :) But well, if it works good enough for you, it is ok. I close the issue so.

closed 2015-03-03 by adegroote