morse-simulator/morse #601: docker image build file available, not exactly an issue with morse itself...

giodegas, 2015-03-14:
Dear morse-fans,

I published a morse docker file:
https://registry.hub.docker.com/u/giodegas/morse/

You can run on any platform, including Windows and Mac Os X, inside a docker image.
Does it make sense?
At the moment the morse check command is happy, but still problems when running it.

Is anybody willing to test it and helping me out to make it work fully?
The docker file is also on github:
https://github.com/giodegas/docker-morse

Thank's.

giodegas, 2015-03-14:
this is my problem now:
* Blender emitted errors during launch:
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory

ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4720:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such file or directory

adegroote, 2015-03-15:
blender seems to support the -noaudio option to disable audio stuff, which is mostly unused in morse. So you can try

```
morse run <your_scene> -noaudio
```

Docker seems to be an interesting platform through I never didn't get time to investigate more on it. Three important questions you may answer:
- do the docker instance can fully access the gpu ? Or do we need to rely on software opengl ?
- how can we interact with docker to select which scene we want to launch ?
- how easy it is to interact with outside world (outside of the docker instance) (through socket / yarp / ros middleware ...) ?

giodegas, 2015-03-16:
ok thank for your hints.
I tried with -noaudio option. but I get:

cd /usr/src/morse
morse run examples/tutorials/tutorial-1-ros.py -noaudio
* Loading simulation </usr/src/morse/examples/tutorials/tutorial-1-ros.py>.

* morse 1.2.2-dirty

* Blender emitted errors during launch:
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4720:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such file or directory

giodegas, 2015-03-16:
also about your question, I am willing to test it via a X session first, with MESA GL.
Than I will try to optimize it, using the GPU if possible.
No problem at all to communicate with the rest of the world by socket, or whatever is needed, as far the used ports are properly handled.

pierriko, 2015-03-16:
You might be interested in our travis script:
https://github.com/morse-simulator/morse/blob/master/.travis.bash

basically the `blender -setaudio NULL` part.

pierriko, 2015-03-17:
update and try
```
morse --noaudio run script.py
```

giodegas, 2015-03-17:
I update the image, but..
    
    morse -v
    morse 1.2.2-dirty
    morse --noaudio run tutorials/tutorial-1.py
    usage: morse [-h] [-c] [--reverse-color] [-v]
             {create,import,rm,add,check,run,edit} ...
    morse: error: unrecognized arguments: --noaudio

why?

giodegas, 2015-03-17:
the git command in the docker file is:

    WORKDIR /usr/src
    RUN git clone https://github.com/morse-simulator/morse -b 1.2.2

adegroote, 2015-03-17:
On 17/Mar - 10:36, Giovanni De Gasperis wrote:
> the git command in the docker file is:
> 
>     WORKDIR /usr/src
>     RUN git clone https://github.com/morse-simulator/morse -b 1.2.2

You need master branch to get its commit

-- 
Arnaud Degroote
ISAE DISC

giodegas, 2015-03-18:
well I tried with master branch, using these commands:

    WORKDIR /usr/src
    RUN git clone https://github.com/morse-simulator/morse -b master

but after the docker build I get:

    morse --version
    morse 1.2-186-gc0fc-dirty

pierriko, 2015-03-18:
the `morse 1.2-186-gc0fc` looks correct,
I don't know why it's `-dirty` though

giodegas, 2015-03-18:
and then:

    morse --noaudio check
    * Checking up your environment...
    * Running on Linux. Alright.
    * Found MORSE libraries in '/usr/local/lib/python3.4/site-packages/morse/blender'. Alright.
    * Trying to figure out a prefix from the script location...
    * Default scene found. The prefix seems ok. Using it.
    * Setting $MORSE_ROOT environment variable to default prefix [/usr/local/]
    * Checking version of /opt/blender/blender-2.73-linux-glibc211-x86_64/blender... Found v.2.73.0
    * Blender found from $MORSE_BLENDER. Using it (Blender v.2.73.0)
    * Checking version of Python within Blender /opt/blender/blender-2.73-linux-glibc211-x86_64/blender... Found v.3.4.2
    * Blender emitted errors during launch: 
    AL lib: (EE) ALCplaybackOSS_open: Could not open /dev/dsp: No such file or directory
    * Blender and Morse are using Python 3.4.2. Alright.
    * Your environment is correctly setup to run MORSE.

and

    morse --noaudio run tutorial-1.py
    * Loading simulation </usr/src/morse/examples/tutorials/tutorial-1.py>.
    * morse 1.2-186-gc0fc-dirty
    * Blender emitted errors during launch:
    AL lib: (EE) ALCplaybackOSS_open: Could not open /dev/dsp: No such file or directory

giodegas, 2015-03-18:
btw, inside the docker image, the files morse/bin/morse.bin includes the code 

    if noaudio:
        other_params += ["-setaudio", "NULL"]

so it should work..

giodegas, 2015-03-18:
the news is that the option is recognized, but blender is still looking for a sound card...
I could eventually investingate how to run it under subuser
https://github.com/subuser-security/subuser
which allows application profiles with X and sound...

pierriko, 2015-03-18:
as a workaround, can you try the travis solution
https://github.com/morse-simulator/morse/blob/master/.travis.bash

```sh
# Avoid ALSA errors on system without sound card
echo -e "#! /bin/sh\nblender -setaudio NULL \$@" > blender
chmod +x blender
export MORSE_BLENDER=$(pwd)/blender
```

some blender args are run sequentially (order matters) so the option might be ignored.
(I check on my desktop and with this option no sound are played, but blender might still look for card)

pierriko, 2015-03-20:
let us know if it sloved your issue

giodegas, 2015-03-20:
well, we can close the issue I think. 
Concepts are clear, even if more testing is needed to be done.
I will eventually open a more specific issue, if needed.
Thank you

adegroote, 2015-03-23:
Close it, as asked by op.

closed 2015-03-23 by adegroote

giodegas, 2015-05-26:
news: I made it work with X (almost) with this new docker image: [giodegas/morse-xvfb](http://registry.hub.docker.com/u/giodegas/morse-xvfb/) with the command:

    docker run -it  -e DISPLAY=$DISPLAY giodegas/morse-xvfb /bin/bash

resulting in an acceptable frame rate.
Just one more fix...
Please have a look at the [morse output log](http://gist.github.com/giodegas/7273e6fda6479ebf9781) and a morse [screenshot](http://picpaste.com/Schermata_2015-05-27_alle_00.06.58-YpHCoNa2.png) .
Why the background is all black?

commit bdbd058f (2015-03-17): [bin] add morse --noaudio CLI option fix #601