morse-simulator/morse #604: Pocolibs issues in some components

lesire, 2015-03-27:
Viam and Stereopixel components still use the old Pocolibs+AbstractDatastream class instead of PocolibsDataStream and PocolibsDataStreamOutput classes.

Results in:
[helpers.loading] Module not found: cannot import name 'Pocolibs'
[helpers.loading] Could not load the class ViamPoster in morse.middleware.pocolibs.sensors.viam
[helpers.loading] Could not create an instance of morse.middleware.pocolibs.sensors.viam.ViamPoster

I can try to make a hack, but as I am not a Pocolibs user, would be better to be nicely fixed by a Pocolibs user :)

adegroote, 2015-03-27:
Do you have BUILD_POCOLIBS_VIAM_SUPPORT enabled in Morse (or viam option in robotpkg) ?

lesire, 2015-03-27:
Yes. Viam is installed, the problem comes from the MORSE viam python 
support who cannot work (uses Pocolibs class which does not exist anymore)

adegroote, 2015-03-27:
Which version of Morse do you use ? 5dc6c545c9abe040cd6cee9a401a3c2fe33ca95c should have fixed it (but only in master branch).

lesire, 2015-03-27:
> Which version of Morse do you use ? 5dc6c54
> should have fixed it (but only in master branch).

Argh... I have installed robotpkg version...

adegroote, 2015-03-30:
Close it, as it is fixed in master.

closed 2015-03-30 by adegroote