morse-simulator/morse #605: is morse good for me ?

user597, 2015-04-02:
Hi 

I want to know if morse is a good choice for my project. I will build a navigation system for vehicles, So, I need to simulate GPS, gyroscope and accelerometre with their error (noise) to check performances of algorithms which  I'll use.
for programming language perhaps I'll use c/c++ so I want to know what kind of input/output your simulator gives, socket or files ? 
If it's not the appropriate for my project, can you please  tell me which alternative I can use ?
thank you

adegroote, 2015-04-03:
Not really an issue per se. The mailing-list morse-users@laas.fr is probably better to discuss about it.

To answer your question, I don't see any issue to use Morse for such scenario, these sensors are simulated, you can add the noise function you want on it. The output can be textual, socket (json), or some robotic middleware such as ROS, YARP, MOOS, ...

user597, 2015-04-03:
thank you :+1:

closed 2015-04-03 by user597

user597, 2015-04-04:
I have another question :
Is it possible to write a motion scenario via a direct contact with actuators (python objects), because when I use socket I don't know exactly the moment at which time simulation start in-order to launch my script , so I can't be sufficiently precise in my simulation.

adegroote, 2015-04-07:
On 04/Apr - 02:29, user597 wrote:
> I have another question :
> Is it possible to write a motion scenario via a direct contact with actuators
> (python objects), because when I use socket I don't know exactly the moment at
> which time simulation start in-order to launch my script , so I can't be
> sufficiently precise in my simulation.

You can add what you want relatively easily. You can just add a new
actuator which does exactly what you want. See for example chapter 4 of

https://www.openrobots.org/morse/doc/stable/user/advanced_tutorials/a_journey_to_a_new_simulation.html

Through, I'm not sure it will be really "more" precise than the socket
option: you can safely assume that, on localhost, transmission is
instant, and so the code will be executed at the next game engine step.