morse-simulator/morse #606: IMU doesn't shows correct values

user597, 2015-04-03:
Hi 

I use IMU instead of accelerometer and gyroscope in-order to have gravity effect. But I noticed that IMU values don't change during motions. so I added to my robot the 3 sensors   accelerometer, gyroscope, and IMU, and compared them 
I used this code : 
```python

imu = IMU()
imu.add_stream('text')
imu.translate(0, 0, 0)
imu.rotate(0, 0, 0)
robot.append(imu)
acc = Accelerometer()
acc.add_stream('text')
acc.translate(0, 0, 0)
acc.rotate(0, 0, 0)
robot.append(acc)
gyr = Gyroscope()
gyr.add_stream('text')
gyr.translate(0, 0, 0)
gyr.rotate(0, 0, 0)
robot.append(gyr)
```
As example, I don't understand why the first and the second elements in linear_acceleration vector are 0 while in accelerometer (-0.0308990478515625, 0.49902331084012985)
ACC :
>==> Data at X,Y,Z: [1.266548 0.004998 0.079990]yaw,pitch,roll: [0.075000 -0.000000 0.000000] | index 16 | time 1428094618.33
>	timestamp = 1428094618328.791260
>	distance = 0.016667
>	velocity = [0.9977817535400391, 0.06661752238869667, 0.0003419816493988037]
>	acceleration = [-0.0308990478515625, 0.49902331084012985, -0.011587142944335938]
IMU :
>==> Data at X,Y,Z: [1.266548 0.004998 0.079990]yaw,pitch,roll: [0.075000 -0.000000 0.000000] | >index 16 | time 1428094618.33
>	timestamp = 1428094618328.791260
>	angular_velocity = Vector((0.0, 0.0, 9.98245468508685e-07))
>	linear_acceleration = Vector((0.0, 0.0, 9.789732933044434))
to see all values  :
Gyro
http://paste.debian.net/164797/
Accelero
http://paste.debian.net/164796/
IMU
http://paste.debian.net/164798/

Is IMU buggy, or I misunderstand how it works ?

user597, 2015-04-04:
I think it didn't work because I used arrows   
```python
keyboard = Keyboard()
robot.append(keyboard)
keyboard.properties(ControlType = 'Position')
```
but it worked when I used  MotionVW  through telnet, 
```bash
id1 atrv.motion set_speed [2, -1]
```

adegroote, 2015-04-07:
On 04/Apr - 00:55, user597 wrote:
> I think it didn't work because I used arrows
> 
> keyboard = Keyboard()
> robot.append(keyboard)
> keyboard.properties(ControlType = 'Position')
> 
> but it worked when I used MotionVW through telnet,
> 
> id1 atrv.motion set_speed [2, -1]

To complete your analysis:

there is basically two (three if we consider force) ways to control a
body in game engine: 
  - velocity control : in this mode, we set velocity and the engine
	computes new position
  - position control : in this mode, we set directly position, and so
	the engine does not know what is really the speed of the body.

and so there is different ways to compute the velocity. First take the
velocity returned by the game engine, or take the position and divide by
time elapsed. Of course, the first option does not really work if you
control your body in position.

Unfortunately, in the stable branche, IMU always compute with velocity,
as long as the robot has dynamic properties. And so, it just does not
work in you control it with a Position Control. In the main branch,
since

733e91386020b064afedb2aaa75de35f6cbf6683,

you should be able to select how IMU computes velocity using the
ComputationMethod property. So I guess the issue is fixed in master.

adegroote, 2015-04-07:
Fixed in master.

closed 2015-04-07 by adegroote