morse-simulator/morse #607: a bug in IMU

abdellani, 2015-04-04:
Hello

I think that I found a bug. The error is in the IMU. When a client ask for "imu.get_local_data()", json.dumps () can't handle the two arrrays "angular_velocity" and "linear_acceleration" stored in vectors <mathutils.Vector>.

	Traceback (most recent call last):
	  File "/usr/lib/python3/dist-packages/morse/middleware/socket_request_manager.py", line 198, in main
	    return_value = json.dumps(r[1][1])
	  File "/usr/lib/python3.4/json/__init__.py", line 230, in dumps
	    return _default_encoder.encode(obj)
	  File "/usr/lib/python3.4/json/encoder.py", line 192, in encode
	    chunks = self.iterencode(o, _one_shot=True)
	  File "/usr/lib/python3.4/json/encoder.py", line 250, in iterencode
	    return _iterencode(o, 0)
	  File "/usr/lib/python3.4/json/encoder.py", line 173, in default
	    raise TypeError(repr(o) + " is not JSON serializable")
	TypeError: Vector((-2.4954491095741105e-07, 5.3199414651317056e-08, 9.034081122072379e-11)) is not JSON serializable

I tried to change the last two lines on /usr/lib/python3/dist-packages/morse/sensors/imu.py

	# Store the important data
	self.local_data['angular_velocity'] = rates
	self.local_data['linear_acceleration'] = accel
 
by

	# Store the important data
	self.local_data['angular_velocity'] = list(rates)
	self.local_data['linear_acceleration'] = list (accel)

It works but I don't if it makes problems in others sides. Also, convertion is approximative :
"angular_velocity" <Vector (0.0000, 0.0000, 0.0000)> <===> [2.3067012477895332e-07, 1.6216743503605358e-08, 5.188650550880247e-11]
"linear_acceleration" <Vector (0.0009, -0.0000, 9.8000)> <===> [0.0009289527079090476, -1.0884766652452527e-06, 9.800002098083496]

I used the following code to test : 
scenario file:

	from morse.builder import *
	atrv = ATRV()
	motion = MotionVW()
	atrv.append(motion)
	imu = IMU()
	imu.add_service('socket')
	atrv.append(imu)
	env = Environment('indoors-1/indoor-1')
	env.set_camera_location([5, -5, 6])
	env.set_camera_rotation([1.0470, 0, 0.7854])

client file:

	import pymorse
	with pymorse.Morse() as simu:
		print (simu.atrv.imu.get_local_data())

adegroote, 2015-04-07:
It should be fixed with d8ab499641058d847e240fc8fe21414404b80aca, but unfortunately, it is not in 1.2_STABLE branch.

adegroote, 2015-04-07:
Fixed in master.

closed 2015-04-07 by adegroote