morse-simulator/morse #612: can't parse get_local_data with json

user597, 2015-04-14:
hello 
I use IMU through socket, but when I try to decode "get_local_data" using json, it bugs
```python
#!/bin/python3
import pymorse
import json
with pymorse.Morse() as simu:
    while True :
        response = simu.atrv.imu.get_local_data()
        print (response)
        list=json.loads(str(reponse))
```
it results
```bash
{'linear_acceleration': [0.0009292324539273977, -7.714302086014868e-08, 9.800000190734863], 'timestamp': 1429020840913.3403, 'angular_velocity': [-1.7000343177642208e-07, 1.7620740777601895e-07, 9.129726835643837e-11]}
Traceback (most recent call last):
  File "recepteur.py", line 23, in <module>
    list=json.loads(str(reponse))
  File "/usr/lib/python3.4/json/__init__.py", line 318, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.4/json/decoder.py", line 343, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.4/json/decoder.py", line 359, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)
```

adegroote, 2015-04-14:
It is not supposed to work :)

The right thing to do is:

```
with pymorse.Morse() as simu:
    while True :
        response = simu.atrv.imu.get_local_data().result()
        # use response as a dictionary
```

Long answer:

get_local_data() returns a MorseFuture (because it can be an asynchronous request). You need to ask for result() to get the real computed value (and it can block in the case of asynchronous request).

Calling print on it just call __repr__ which itself call result() and then print the dictionary (but even if python dict looks like JSON, they are not JSON hence the error).

Please confirm that it works and close the issue if so.

user597, 2015-04-14:
Yes, it works... 
```python
        reponse = simu.atrv.pose.get_local_data()
        tmp=reponse.result()
        print (tmp['timestamp'])
```

closed 2015-04-14 by user597