morse-simulator/morse #613: use a OSM map as environment

user597, 2015-04-15:
Hello 

I used this addon https://github.com/vvoovv/blender-geo/wiki/Import-OpenStreetMap-(.osm) to import OSM maps on blender. I saved the file and used as environment for my scenario. But from the begging the robot (atrv in my case) start a free fall, and disappear for the view. is there a way to stop this free fall, and to make the car run on streets ?

adegroote, 2015-04-15:
Without the scene you test, it is difficult to answer precisely. Do your ground is physically static ?
Consider "car run on the streets", there is some path finding algorithm in Blender, with navigation meshes, but there is nothing specific done in Morse.

user597, 2015-04-15:
>Do your ground is physically static ?
after loading the map, streets are shown as lines, So I guess I need to replace lines with 3d streets ?

adegroote, 2015-04-15:
Without more information about your blend file, it is really hard to say anything. But yes, you need some plane if you want to put any body on it.

giodegas, 2015-05-25:
it has to be a plane, or can it be a terrain-like topological surface?

adegroote, 2015-05-25:
On 25/May - 03:10, Giovanni De Gasperis wrote:
> it has to be a plane, or can it be a terrain-like topological surface?

it can be whatever you want, but it needs to have a physics existence,
i.e. being of type 'static'.

maccradar, 2015-12-07:
Did you make progress with your OSM import because we have similar interests? Did the georeferencing work in Morse after the import in Blender and how did you make 3d streets out of the lines in the OSM map?