morse-simulator/morse #614: getting absolute position using socket

user597, 2015-04-16:
Hi
is it possible to get absolute position through socket, like what you find in X,Y,Z and yaw,pitch,roll vectors  when you using text data stream  ?
```bash
==> Data at X,Y,Z: [0.004845 -0.004483 0.103543]yaw,pitch,roll: [0.913732 -0.000095 -0.000000] | index 39 | time 1429093673.41
        timestamp = 1429093673414.843750
        angular_velocity = Vector((3.3109475339188066e-07, 6.889659687203675e-08, -1.0246163917315698e-09))
        linear_acceleration = Vector((0.0009289041627198458, 1.246240799446241e-07, 9.800000190734863))
```

adegroote, 2015-04-16:
Adding a Pose Sensor https://www.openrobots.org/morse/doc/latest/user/sensors/pose.html will give you this information.

Please don't use the issue system for this kind of question, but the mailing-list morse-users@laas.fr.
Moreover, you can take a look at the list of components

https://www.openrobots.org/morse/doc/stable/components_library.html

to show exactly what kind of sensors / actuators is available and which information it returns.

closed 2015-04-16 by adegroote

user597, 2015-04-16:
ok, thank you ... 
I'll never repeat it again