morse-simulator/morse #619: MORSY behaviour is unrealistic

severin-lemaignan, 2015-04-21:
Using the new 'sandbox' environment, and driving Morsy onto the ramp, the physical behaviour of the robot is completely unrealistic and give a bad first impression of MORSE.

The physical behaviour of Morsy needs to be fixed.

Steps to reproduce:
```
$ morse create test & morse run test
```
then, drive Morsy to the ramp with the keyboard: the robot alsways remains horizontal, and does not fall out from the ramp properly.

adegroote, 2015-04-21:
Can you test https://github.com/adegroote/morse/tree/fix_619 and tell if it looks better for you.

severin-lemaignan, 2015-04-21:
Well, that's certainly better, but still not very good... the robot seems to 'fly' a little bit sometimes, it does not always stabilize after a fall...

adegroote, 2015-04-23:
I don't know if I we can do a lot better without heavy modification. 
I think the "first" issue is that Morse always executes "something" in the actuator (the last values it gets from the middleware or the default value). And  executing v = [0.0, 0.0, 0.0] is not the same thing that doing nothing (basically it works against the gravity). 

Moreover, it is not possible to control only one direction in Blender, you need to control the full velocity vector. The "ground robot" stuff try to mitigate this issue telling basically that we cannot control vz / rx / ry.

The second issue is that Morse keyboard and v_omega actuator controls the robot in robot frame. And so, when Robot_x points  towards z, controlling vx makes the robot "flying". In this setup, the "ground robot" control is not correct at all...

adegroote, 2015-04-23:
Finally I get some idea / better understanding. Can you test the current version of my branch ?

severin-lemaignan, 2015-04-23:
Thanks for looking into it!
It seems better (so I would merge it in any case -- maybe corresponding unittests would be useful, though), but still not very good.

Cf video: https://www.youtube.com/watch?v=hSbwAgpSkfQ

adegroote, 2015-04-24:
For the part "the robot is not always standing vertically", I can't do more I think for a "cube"  flying over a few cm over the ground. It looks like a bit a "segway" so not so bad. It 'worked' before because the robot was "dynamic" against "rigid body".

For "falling" part, I don't know exactly what you want to have as behaviour. But I think the issue is not in the code, but in the Morsy model. It probably should have a center of mass lower on the robot.

severin-lemaignan, 2015-04-24:
Alright, I've made another, better video with physics vizualistion enabled + display of keyboard strokes.
We still have issues that are not only related to the physical model of the robot...
https://www.youtube.com/watch?v=J3mQrzgHnAQ

adegroote, 2015-04-24:
Can you precise which part of the video seems "strange" for you ?

severin-lemaignan, 2015-04-24:
Well, all :-)
First (but it is expected based on your changes, the robot remains horizontal while the key are pressed, and start stabilizing when I release the key (see when I go up the ramp, or when I turn on the ramp).
Then, when back to the ground, the robot remains in a weird, tilted pose, even when I do not press any key.

severin-lemaignan, 2016-01-11:
Closing for now.

closed 2016-01-11 by severin-lemaignan