morse-simulator/morse #620: CMake warning: dependency hla/omt.py of 'copy_doc' does not exist

severin-lemaignan, 2015-04-23:
Doing a clean install of the current trunk, CMake gives the following warning (knowing that I did not enabled any middleware in CMake):

```
CMake Warning (dev) at CMakeLists.txt:219 (ADD_DEPENDENCIES):
   Policy CMP0046 is not set: Error on non-existent dependency in
   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
   Use the cmake_policy command to set the policy and suppress this warning.

   The dependency target "/home/skadge/src/morse/build/fakeenv/hla/omt.py" of
   target "copy_doc" does not exist.
 This warning is for project developers.  Use -Wno-dev to suppress it.
```

adegroote, 2015-04-24:
Which version of cmake ? How do you call it ? Don't have this issue on 2.8.12.2 with cmake ../.

severin-lemaignan, 2015-04-24:
Tested on Debian Jessie (ie `cmake-3.0.2`): this may well be the reason!

adegroote, 2015-04-24:
Can you check it really improve thing on 3.0.2. Works properly on 2.8.12.2.

severin-lemaignan, 2015-04-26:
This fixes the issue for me. You can merge it. Thank you!

commit efcbb4d7 (2015-04-27): [cmake] Please cmake-3.02 and simplify a bit

Fix #620
closed 2015-04-27 by adegroote