morse-simulator/morse #621: Install by default support for all middleware that do not require compilation

severin-lemaignan, 2015-04-27:
What is the reason not to install ROS and MOOS support by default (they are Python only, ie they do not require header or library to install)?

Installing them would provide a much better 'out of the box' experience, especially for ROS:  people are used to install ROS via the Debian package, they then install morse with the Debian package as well, to finally see that... it does not work! It gives a bad first impression, I think.

adegroote, 2015-04-27:
Basically, all middleware are python only except pocolibs viam/stereopixel/velodyne extension. 

Of course, we can remove all check in cmake, just install everything, and let user deals with dependencies.

 From a point of view of package maintainers, it is probably a bit messy, because maintainers expects that dependencies are properly packaged (which is the reason why we separate stuff properly). 
The "right" solution would be to handle #125 in some ways :). 

With the current implementation, you can ask debian maintainer to just include ROS in the default build, as there is no check for it.

An in-between solution is to remove the check but keep the different options (OFF by default). A maintainer could choose to just create one package with all-dependencies or keep the package splitted, as done for the moment.

severin-lemaignan, 2015-04-28:
And what if we keep the options ON by default, so that packagers can still properly package if they want to, but the end-users have a good experience out of the box?
I guess it is a matter of giving priority to packaging correctness vs end-user experience...

adegroote, 2015-04-28:
It is an option. The only drawback is for developer  which runs tests, (so a set of 1 person :)) which need to disable explicitly non-needed middlewares.

If nobody objects, we could do that. Can you take care about doc upgrade ? I will take care about cmake stuff.

adegroote, 2015-04-28:
See my branch https://github.com/adegroote/morse/tree/fix_621 for the cmake stuff and few other related commits

severin-lemaignan, 2015-04-28:
One issue with your patch:
If you disable all options except `BUILD_DOC_SUPPORT`, you get the following CMake warning:
```
 CMake Warning (dev) at CMakeLists.txt:151 (ADD_DEPENDENCIES):
   Policy CMP0046 is not set: Error on non-existent dependency in
   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
   Use the cmake_policy command to set the policy and suppress this warning.

   The dependency target "zbufferto3d" of target "generate_doc" does not
   exist.
 This warning is for project developers.  Use -Wno-dev to suppress it.
```
(with another one for `zbuffertodepth`)

adegroote, 2015-04-29:
a7c9347 should handle the last reported warning.

adegroote, 2015-04-29:
Merged.

closed 2015-04-29 by adegroote