morse-simulator/morse #624: Camera to Joint

fdevillalobos, 2015-05-27:
Hi... I looked through all the issues and tutorials and cannot seem to find an answer for this.

I have a tripod with a camera on top of it. The camera is mounted in a devise that can pan and tilt.
I've done that... created the joints, and I'm able to pan and tilt the OBJECT in the robot to the position that I want to... but NOT the camera.
Is there any way I can append a camera instead of to a robot, to a joint of the robot??
I'm importing the whole geometry as one blender file with everything on it, and then I use actuators to pan and tilt the parts that I need to... but it's being a pain to move the camera... can the camera just follow a part of the robot I tell it to?

Thanks,
Francisco.

adegroote, 2015-05-28:
For a pan-tilt purpose, you can look at the PTU actuator, and its unit-test 

https://www.openrobots.org/morse/doc/stable/_modules/base/ptu_testing.html

(the gyroscope is on the movable pan tilt unit).

For a more general purpose of adding a component not to the robot, but to a specific subobject of the robot, there is no particular syntax (nowadays at least but it may make sense to add an optional argument to append) but you can try the following

```
atrv = ATRV()

camera = VideoCamera()
atrv.append(camera)

wheel = atrv.get_child('Wheel.1')
camera._bpy_object.parent = wheel
```

fdevillalobos, 2015-05-28:
Thanks so much for this! It really helped. It was just what I needed.

Just one more quick question and I may have it solved. How can I go further into the child path??
If I do atrv.get_child('Wheel.1').get_child('name1') it doesn't work...

I here attach an image of my object tree. I need to get all the way down to 'Ernie'!

[image]

I tried several syntax's and couldn't figure it out. Thanks for the help in advance! Really useful!

Francisco.

adegroote, 2015-05-28:
On 28/May - 09:37, Francisco de Villalobos wrote:
> Thanks so much for this! It really helped. It was just what I needed.
> 
> Just one more quick question and I may have it solved. How can I go further into the child path??
> If I do atrv.get_child('Wheel.1').get_child('name1') it doesn't work...
> 

It is expected that it does not work, get_child() is a method of
AbstractComponent, but returns a blender object, so you can't chain
them.

However, get_child searches recursively, so normally, fred.get_child('Ernie')
should work.

fdevillalobos, 2015-05-28:
Thanks so much! That works perfectly! Very helpful!

closed 2015-05-29 by adegroote