morse-simulator/morse #625: VideoCamera Image Size

fdevillalobos, 2015-06-03:
Hello there,

I was wondering if I'm doing anything wrong while trying to read the image of a video camera.
In the documentation it's said that the buffer of video[image'] should be (cam_width * cam_height * 4).
I always get it bigger. For example in a 100x100 image I get 53336 instead of 40000.

In a client script I'm just doing something in the lines of:

def video_image(video):
  print(video['timestamp'])
  print("Image size is %s" % len(video['image']))

with Morse() as simu:
    video = simu.robot.videocamera.subscribe(video_image)

And in the printing of the data I'm getting those numbers that I mentioned. I tried several sizes and the length of the buffer seems to be always longer than the buffer should actually be.

Any idea why??
Would also really appreciate how I could do to save this image to disk in python and/or get a live video feed of those images in a separate terminal.

Thanks,
Francisco.

adegroote, 2015-06-03:
Hum, 53336 looks weird, I will expect 65536. If I remember properly, Blender (for OpenGL need) rounds up to the next power of two, so 100 -> 128. 

For live feed, we generally use ROS or YARP middleware which provide a tool for that. There is no such tool for Morse-socket for the moment (through it should not be too hard to implement). Anyway, there is some example how to dump an image grabbed from socket in the video_image unit-test, see

https://www.openrobots.org/morse/doc/stable/_modules/base/video_camera_testing.html#save_pgm_ascii

fdevillalobos, 2015-06-04:
Hey,

Thanks for the info. We were able to get out an image in the pgm format.
What I did see was that the image is been decoded and then converted to grayscale. I just wanted to know if we could skip the grayscale conversion, and just get an rgb image out.

Our final output to a message has to be an rgb image as we have the pipeline on the other side already done.
Thanks in advance for the info. You've been really helpful.
Francisco.

adegroote, 2015-06-04:
The greyscale conversion is done by rgba2gray8u, you can skip it, and just use the save_pgm method.

smishraasu, 2015-06-04:
Hi,

When we skip the rgba2gray8u method we get an assertion error : "assert(len(image8u) == width * height)
AssertionError "

adegroote, 2015-06-04:
On 04/Jun - 02:09, smishraasu wrote:
> Hi,
> 
> When we skip the rgba2gray8u method we get an assertion error : "assert(len(image8u) == width * height)
> AssertionError "
> 

You should use save_pgm, not save_pgm_ascii

smishraasu, 2015-06-04:
Hi, 

do you mean the save_image method, because I don't see any save_pgm method.

smishraasu, 2015-06-04:
Also then I try the save_image method, I get the following error "struct.error: required argument is not an integer" . I am calling the save_image method in capture8u method as follows : 

def capture8u(cam_stream, image_path=None):
    capture = cam_stream.get()
    image8u = capture['image']
    save_image(image_path,image8u)

and I have a hardcoded image path for test purposes and the extension is png.

adegroote, 2015-06-19:
See f746c5d094ace4f8eb5ea4666afce3ecf013af08 for some other examples about dumping image from socket flow. Can I close the pr ?

smishraasu, 2015-06-19:
Sure @adegroote  .. Thanks for the help :)

closed 2015-06-22 by adegroote