morse-simulator/morse #629: Yosemite and 1.3

fdevillalobos, 2015-06-04:
Hello community! Nice to have such support.

So... what is the issue with Yosemite and Morse 1.3?? I thought that for this version I would have no more Python problems trying to run here.
Can anybody point me out on what to do??

In Yosemite, Blender less than 2.72 does not run... and yet, when I try to run morse I get the error:

* Blender is compiled for Python 3.4.2 but MORSE has been compiled for Python 3.3.0! Check your MORSE build configuration or the selected Blender version.

So... does anybody have any tips here? I saw an older post about this but it was already closed.

Thanks!
Francisco.

adegroote, 2015-06-05:
I don't speak enough MacOSX to remember which version is Yosemite exactly :) Through, I do some test recently on MacOSX and it should work, as on other platform. I have modified / improved the documentation for homebrew too.

https://www.openrobots.org/morse/doc/stable/user/installation/package_manager/homebrew_osx.html

Basically, you seem to build Morse with the default Python version installed by MacOSX, while you need a more recent one to run Morse (the same version that it is bundled with Blender).

fdevillalobos, 2015-06-08:
First of all thanks for the quick response!

Yosemite is the latest version of OS X available. I'm having several issues here. Already fixed most of the errors I was getting... now I'm getting one I cannot seem to overcome.

1) If you import any blender model that is parented to a "null" that only has a translation, rotation or anything like that... then the object does not show... but is still there. Like the ground. It did not appear... but it did not allow things to fall off! This has an "easy" fix by deleting all the null objects, and non-mesh objects... but in really complicated, multi-body objects... you will probably have some nulls there. Not a big deal... maybe something to fix...
Same model worked in 1.2 and dissapeared in 1.3 (I'm mainly talking OS X).

2) This is king of a bigger one. I'm getting this weird error when I import new robots that I actually created:Traceback (most recent call last):
  File "/Users/prenav/sbox/fred_sim_1/default.py", line 73, in <module>
    robot = Quad()
  File "/Users/prenav/sbox/fred_sim_1/src/fred_sim_1/builder/robots/quad.py", line 36, in __init__
    self.snapcontroller = Snapcontroller()
  File "/Users/prenav/sbox/fred_sim_1/src/fred_sim_1/builder/actuators/snapcontroller.py", line 7, in __init__
    "snapcontroller")
  File "/usr/local/lib/python3.4/site-packages/morse/builder/creator.py", line 139, in __init__
    make_morseable)
  File "/usr/local/lib/python3.4/site-packages/morse/builder/creator.py", line 66, in __init__
    obj.game.physics_type = 'NO_COLLISION'
AttributeError: 'NoneType' object has no attribute 'game'

What does this NoneType No Collision mean?? I'm putting the blend file with the second error here.
https://www.dropbox.com/s/mhl8ib9g7aandxq/Quad_not_working.blend?dl=0&s=sl

And this is a blend file that will not render on the Viewport.
https://www.dropbox.com/s/e9n10buibg9v773/Bert.blend?dl=0&s=sl

If I could just fix the Quad file working... fixing the error that it's giving me... I would already have a working environment.

Thanks!
Francisco.

adegroote, 2015-06-11:
Can you share the code too ? (not on dropbox please, cannot access it from $DAYJOB).

fdevillalobos, 2015-06-19:
I'm sorry for the silence. Trying not to bug very much. But I found one real issue now. First: The code:

```python
from morse.builder import *

robot = Quadrotor()

gps = GPS()
# gps.level("raw")
gps.translate(x=.1, z = 0.1)
gps.add_stream('socket')
gps.alter( 'UTM' )
gps.alter( 'Noise', pos_std=1.0 )
robot.append(gps)

# Robot Translation and interface
robot.translate(1.0, 0.0, 1.2)
robot.rotate(0.0, 0.0, 3.5)
robot.add_default_interface('socket')

## ENVIRONMENT
# set 'fastmode' to True to switch to wireframe mode
env = Environment('sandbox', fastmode = False)
env.set_camera_location([-18.0, -6.7, 10.8])
env.set_camera_rotation([1.09, 0, -1.14])
```

Simple enough code that should work no problem at all. But the commented out line:
gps.level("raw")
errors the program out like this:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/site-packages/morse/blender/calling.py", line 54, in sensor_action
    component_action(contr)
  File "/usr/local/lib/python3.4/site-packages/morse/blender/calling.py", line 50, in component_action
    cmpt_object.action()
  File "/usr/local/lib/python3.4/site-packages/morse/core/sensor.py", line 76, in action
    self.default_action()
  File "/usr/local/lib/python3.4/site-packages/morse/sensors/gps.py", line 238, in default_action
    gps_coords = self.coord_converter.ltp_to_geodetic(xt)
AttributeError: 'NoneType' object has no attribute 'ltp_to_geodetic'
```

This is happening in morse 1.3, in MacOSX. I tried to compare with Ubuntu 14 and Morse 1.3, but even though I download the 1.3 branch, the source code or whatever I do (even apt-get install morse-simulator), it shows as Morse 1.2 when I launch it... maybe you just did't change the text that appears when you launch morse?? I don't know.
So, this works in Ubuntu, but not in Mac... any ideas why?? Thanks!
Francisco.

adegroote, 2015-06-19:
coord_converter is something I introduced in current, not in 1.3.

Nevermind, the issue is that to be able to use "raw_gps", you need to define longitute, latitude and altitude at the environment level (before (in 1.3 so), it was a property of gps, which some dummy default).

So, add
'''
env.properties(longitude = 1.26, latitude = 43.26, altitude = 130.0)
"""

for example should solve your issue (somewhere near Toulouse, France)

severin-lemaignan, 2015-06-19:
@adegroote In case of missing longitude/latitude/altitude, is the error message clear to the user in the current code?

adegroote, 2015-06-19:
https://github.com/morse-simulator/morse/blob/master/src/morse/helpers/coordinates.py#L40

should warn "relatively clearly" the user, but I can rethrow the exception too, to stop the execution :)

severin-lemaignan, 2015-06-19:
@adegroote If you want to be really gentle :-), add smthg like:
```
logger.error("Missing parameter %s for coordinate conversion: add 'env.properties(%s=...)' to your builder script.\n", (e,e))
```

adegroote, 2015-06-19:
Will add it in my coordinates_numpy branch ;)

adegroote, 2015-06-22:
Concerning the " File "/usr/local/lib/python3.4/site-packages/morse/builder/creator.py", line 66, in init
obj.game.physics_type = 'NO_COLLISION'
AttributeError: 'NoneType' object has no attribute 'game'", it is probably related to some change in ActuatorCreator, you should now have something like

```
class SnapController(ActuatorCreator):
    _classpath = "...."

   def __init__(self, name = None):
       ActuatorCreator.__init__(self, name) 
```

fdevillalobos, 2015-06-22:
Yes! That was what was happening with all the actuators, robots, and sensors I created... Once I runned the add command, everything worked, but not just with the same topic. Will this same thing still work for 1.2 so that I can unify everything?
Or should I have two branches one for MacOSX and one for Linux??
Thanks again!

adegroote, 2015-06-23:
The signature of Creator has changed a lot between 1.2 and 1.3, so I don't think we can have "one" syntax. But you can try something like:

```
from morse.version import VERSION

class SnapController(ActuatorCreator):
     _classpath = "..."

    def __init__(self, name = None):
        if VERSION < "1.3":
            old syntax
        else:
             ActuatorCreator.__init__(self, name)

adegroote, 2015-06-30:
Can I close it ?

fdevillalobos, 2015-06-30:
Yeah... sure. Everything working at the moment!
Though we had some issues with the builder for different versions. Still figuring out some things though we mainly decided to go all the way to 1.3. He.
Thanks!

closed 2015-07-03 by adegroote