morse-simulator/morse #635: Multinode Simulation

smishraasu, 2015-06-30:
Hi, 

I have been using Morse till now and my experience has been really great. Thanks for all the support. 

I have a small question regarding multi-node distribution. 
Suppose I have 2 ATRV robots in the scene. 

In the builder files they are defined as - 

car = ATRV()
jeep = ATRV()

I follow the following steps as described in the link https://www.openrobots.org/morse/doc/stable/user/multinode/socket.html#the-multinode-server-program 
to setup multinode simulation. 

I have two client files corresponding to each ATRV robots, where I have the following lines as a part of  each of the client files : 
from pymorse import Morse
with Morse() as simu:

But it seems that when I run the client corresponding to "car = ATRV" first and then the client corresponding to "jeep=ATRV()" I get the error 'Morse' object has no attribute 'jeep' and when I run the client corresponding to "jeep=ATRV()" first and then the client corresponding to "car=ATRV()" , I get the error 'Morse' object has no attribute "car". So in multinode simulation when one node is handled, it treats the other robots as external. How do you I achieve running two robots from two different scripts, in a scenario where I want to control "car" and "jeep" independently using different keys on keyboard.

The following is what I am doing " 

Terminal 1 - multinode_server
Terminal 2 - export MORSE_NODE=nodeA

adegroote, 2015-06-30:
Can you share all the code that you use, so it is easier to diagnostic the issue ?

smishraasu, 2015-06-30:
Thanks for the quick response. I was actually trying to recreate the scenario and explain the issue that I was facing. I would have definitely shared the code, but I am not able to do that due to confidential reasons. Please let me know if I could explain the scenario better. Sorry again.

adegroote, 2015-06-30:
Do you run both simulation on the same machine ? 
How do you configure your pymorse call ?

If they both run on the same computer, one should be:

```
with Morse('localhost', 4000) as simu:
    ...
```

and the second

```
with Morse('localhost', 4001) as simu:
    ...
```

If they are on different host, you should use port 4000 everywhere, but adapt the host.

smishraasu, 2015-06-30:
You are awesome and that works :D . Great :D . Just one more question, when I run both the simulations it basically opens up two simulations, is it anyways possible to just open up one window and still have a multinode structure. Thanks a lot again :D

adegroote, 2015-06-30:
Unfortunately not at the moment. Morse is currently strongly tied to Blender, and it is not possible to run a simulation without output. Running multinode on one compute does not really make sense in general  (apart for test purpose).

adegroote, 2015-06-30:
Can I close it ?

smishraasu, 2015-06-30:
The reason I was thinking for multinode simulation was because when I had two robots in one simulation it would slow down the frame rate from 60fps to 40fps. But now it seems that when I assign the two robots to two different nodes and run the simulation the frame rate remains almost constant at 60fps with rare drops to 48fps

closed 2015-07-12 by smishraasu