morse-simulator/morse #637: Pioner3DX and PatrolBot make a random rotation before moving forward

kellymorse, 2015-07-07:
Hello,
I am new in morse, and I am using it with ros interface, I am having issues with two robots (Pioner3DX and PatrolBot) when trying to move them forward, backward, left or right, the robot make a very fast random rotation (in many times it is around 90deg.) when receiving the first command (with and without ros interface), witch is very annoying for my SLAM algorithm.
I am using morse 1.2 with ubuntu 12.04 and Groovy ros
Any help please will be appreciated.
My .py file is bellow:

1- keyboard control:
from morse.builder import *
##################### ROBOT  ####################
p3dx0 =   Pioneer3DX()#PatrolBot()
p3dx0.translate(x=-38.0, y=38.0, z=0.0)
##odometry
odom0 = Odometry()
...
##laser 
scan = Sick()
...
## place your component at the correct location
keyboard = Keyboard()
keyboard.properties(Speed=1.0)
p3dx0.append(keyboard)
##################### ENV.  ####################
env = Environment('/myhomeDIR/envs/maze', fastmode=False)
env.place_camera([0, 0, 60])
env.aim_camera([0, 0, 0])

2- ros topic control
motion = MotionVWDiff()
motion.add_interface('ros',topic='/cmd_vel')
p3dx0.append(motion)

adegroote, 2015-07-09:
Can you provide a complete and minimal builder script exhibiting the issue ?
Can you try with 1.3 ?

It is strange that you can move at all the robot using your keyboard setup, you  need at least

```
keyboard.properties(ControlType = 'Differential')
```

to make it working properly.

Can you try to launch the pionner3dx_testing.py test (in MORSE/testing/robots/pionner3dx/) ?

kellymorse, 2015-07-09:
thank you for the help,
I am using morse 1.2, never tried 1.3

adding this line "keyboard.properties(ControlType = 'Differential')"
does not help, I am getting the same behavior!

Yes I am able to launch pionner3dx_testing.py but can't move the robot!?

commit 8a6e319b (2015-07-29): [core] Don't induce a rotation when awaking the wheeled robot

Fix #637
closed 2015-07-29 by adegroote