morse-simulator/morse #638: AttributeError: 'AlwaysSensor' object has no attribute 'frequency'

pringlized, 2015-07-08:
I'm new to Morse.  Installed it recently using the [Homebrew instructions](http://www.openrobots.org/morse/doc/1.3/user/installation/package_manager/homebrew_osx.html) on my MacBookPro running Yosemite 10.10.3.  Running the initial sockets tutorial and also connecting to it from a socket client works fine. I added a few things like a light, and had it just drive in a circle.  No problem.  But when I start adding a variety of sensors it errors out with: AttributeError: 'AlwaysSensor' object has no attribute 'frequency'

I tested a variety of sensors and most fail on creating the object: SemanticCamera, VideoCamera, SickLRDS, Proximity, etc..  But some do work like: Pose, Collision, etc..

Been digging through source code even though not very familiar with Blender's source or Morse.  Haven't been able to nail down the culprit yet.  Any help would be much appreciated.

Great project!  Appreciate all the work that has been done on it. Thanks.

##### Running tutorial code with a Proximity sensor
```
$ morse run tutorial-1-sockets.py 
* Loading simulation </Users/jim/Dev/morse-sim/tutorials/tutorial-1-sockets.py>.

* morse 1.3-dirty

* MORSE has only been tested on Linux. It may work
on other operating systems as well, but without any guarantee

* Version 2.75.0 of Blender is untested but should work

ndof: 3Dx driver not found
found bundled python: /Applications/Blender_2.75/blender.app/Contents/MacOS/../Resources/2.75/python
read blend: /usr/local/share/morse/data/morse_default_autorun.blend
[ERROR][MORSE] Uncaught exception, quit Blender.
Traceback (most recent call last):
  File "/Users/user/Dev/morse-sim/tutorials/tutorial-1-sockets.py", line 17, in <module>
    proximity = Proximity()
  File "/usr/local/Cellar/morse-simulator/1.3/lib/python3.4/site-packages/morse/builder/sensors.py", line 111, in __init__
    self.frequency(12)
  File "/usr/local/Cellar/morse-simulator/1.3/lib/python3.4/site-packages/morse/builder/abstractcomponent.py", line 523, in frequency
    morselogic[0].frequency = delay
AttributeError: 'AlwaysSensor' object has no attribute 'frequency'
Traceback (most recent call last):
  File "/Users/user/Dev/morse-sim/tutorials/tutorial-1-sockets.py", line 17, in <module>
    proximity = Proximity()
  File "/usr/local/Cellar/morse-simulator/1.3/lib/python3.4/site-packages/morse/builder/sensors.py", line 111, in __init__
    self.frequency(12)
  File "/usr/local/Cellar/morse-simulator/1.3/lib/python3.4/site-packages/morse/builder/abstractcomponent.py", line 523, in frequency
    morselogic[0].frequency = delay
AttributeError: 'AlwaysSensor' object has no attribute 'frequency'
```

##### Running tutorial WITHOUT Proximity sensor
```
$ morse run tutorial-1-sockets.py 
* Loading simulation </Users/user/Dev/morse-sim/tutorials/tutorial-1-sockets.py>.

* morse 1.3-dirty

* MORSE has only been tested on Linux. It may work
on other operating systems as well, but without any guarantee

* Version 2.75.0 of Blender is untested but should work

ndof: 3Dx driver not found
found bundled python: /Applications/Blender_2.75/blender.app/Contents/MacOS/../Resources/2.75/python
read blend: /usr/local/share/morse/data/morse_default_autorun.blend
morse.sensors.collision.Collision: no interfaces available for this component! Trying to use default one for socket.
Detected GL_ARB_texture_env_combine
Detected GL_ARB_texture_cube_map
Detected GL_ARB_multitexture
Detected GL_ARB_shader_objects
Detected GL_ARB_vertex_shader
Detected GL_ARB_fragment_shader
Detected GL_ARB_vertex_program
Detected GL_ARB_depth_texture
Detected GL_EXT_separate_specular_color

Blender Game Engine Started
==============================
[... 210 lines cut ...]
env.set_camera_rotation([1.0470, 0, 0.7854])
```

##### Client code
```python
from pymorse

def pose_received(pose):
    print("The Robot is currently at %s" % pose)

def main():

    with pymorse() as morse:
        morse.atrv.pose.subscribe(pose_received)
        motion = morse.atrv.motion

        v = 1.5
        w = 1

        while True:
            motion.publish({"v": v, "w": w})

if __name__ == "__main__":
    main()
```

pringlized, 2015-07-08:
Can't believe I missed this in the PRE-INITIALIZATION:

```
Method SCA_ISensor.frequency is deprecated, please use SCA_ISensor.skippedTicks instead.
/usr/local/Cellar/morse-simulator/1.3/lib/python3.4/site-packages/morse/core/object.py:56
```

I'm guessing that maybe my version of Blender isn't using SCA_ISensor.frequency any longer?

adegroote, 2015-07-08:
Yes, it is a change in blender API in 2.75. This is handled by a0364ae2ec3f819282b1d5037d7ec01483fcc83f in master and 54f1b72c0ab4291653daf394c0b5267dc5aead58 in 1.3_STABLE. But the release 1.3 does not have the patch

pringlized, 2015-07-09:
Thank you. Feel free to close the issue.

closed 2015-07-09 by adegroote