morse-simulator/morse #639: async capture service for videocamera

smishraasu, 2015-07-12:
I have a videocamera attached to a groundrobot. And in the client file I have the following : 

def print_image(video):
  print(video['image'])

res = simu.fredbot.ptu.fredview.capture(1)
while res.running():
   video = simu.fredbot.ptu.fredview.subscribe(print_image)
   simu.sleep(0.5) 

At times it prints one image and at times it doesn't print anything to the terminal. Could you tell me what could be the problem, because of which it is not subscribing to the captured image at times. 

Thank You.

smishraasu, 2015-07-12:
It seems that whenever I am trying to capture exactly one image, it doesn't capture it, when I am trying to capture 2 images, it captures 1 and so on.

adegroote, 2015-07-13:
It is possible there is an off-by-one computation in capture service. Through, I think your test code is incorrect (and it is a bit hard to guess what the code does really as it is not indented properly). 

First, suscribe can be called once (it basically register the callback so no need to call it more than once). Moreover, you need to call it before the capture call, otherwise, there is a possible race condition (the data is captured before the suscribe is done). 

Can you confirm you still see the issue after fixing the previous points ? If it is the case, I will investigate more on it on Wednesday.

smishraasu, 2015-07-16:
Thank you again @adegroote  for the support. I made the changes were, the subscribe is called once and then I call the capture asynchronous service. It works great. 

Thank You. I am closing this issue.

closed 2015-07-16 by smishraasu