morse-simulator/morse #641: Improve support of Morse on Windows

adegroote, 2015-07-15:
This is a meta-bug to report what is missing on Windows side, or portability issue. 

A first important issue is that "morse run xxx" does not work properly on Windows, while morse edit xxx then P works.

nicolaje, 2015-08-05:
Regarding the "morse run xxx" issue, I can say it was introduced from Blender 2.68, as it worked fine on Windows until Blender 2.67a.
I think the bug was introduced in https://developer.blender.org/rB412c04347440fa5e480b69a07b20ff1399775c5b and I filled a bug report on the Blender Bug Tracker: https://developer.blender.org/T45699?workflow=create
Let's see how it goes

nicolaje, 2015-10-23:
The bug was corrected:
https://developer.blender.org/T45699
There is still a small one that pauses the game engine when the window is moved, but it is much better now. Now windows support is good from the Blender perspective.