morse-simulator/morse #646: Issue with robots in blender versions >2.69

fdevillalobos, 2015-08-05:
Hello community,

I've been having a problem adding new robot meshes to morse in new blender versions.
This happens ony in blender versions newer than 2.69.
I can open it and do things normally in blender with the model... but when I open the morse simulation, I go down from 60fps to 0.3fps just by changing a small thing in the blend file. Can anybody help with this??

I'm attaching here two very similar files... one works fine, and one works really bad in any simulation you put in to. I tried empty environments... everything. Any ideas??

Also when I go to blender 2.75a, the whole system crashes before it even starts which I don't care because I'm using blender 2.71.

Thanks a lot,
Francisco.
[image]

[image]

adegroote, 2015-08-06:
For 2.75, do you have the patch https://github.com/morse-simulator/morse/commit/54f1b72c0ab4291653daf394c0b5267dc5aead58 ? (it is not in the release, but it is in 1.3_STABLE).

I will try to look at the issue with your blend file this afternoon, but after that, I'm going in vacation, so it will wait a few weeks.

fdevillalobos, 2015-08-07:
Yeah... I'm already on 1.3_STABLE and I checked for those lines to be there and they are... so that's not a fix for this particular problem. Let me know if you find something... and good luck on your holidays!!

adegroote, 2015-08-26:
From a quick check, the difference is that the "bad" model has an extra tilt_axis with some physics "dynamic".  Moving it from dynamic to "No Collision" makes the two models react in the same way.

Of course, I don't know what you are trying to do, but having a "physical body" into another one is not really a good idea, and can make the physical engine do a lot of interesting stuff.

adegroote, 2015-08-26:
For 2.75, does it happen on any scene or only on some of your specific scenes ? In the second case, can you share one ?

adegroote, 2016-01-05:
Does it work with 2.76 ? Can you share a "non-working" scene ? If no answer in the two next weeks, I will close the issue.

adegroote, 2016-01-12:
Close it, as we don't have more information

closed 2016-01-12 by adegroote