morse-simulator/morse #649: Noise in accelerometer

minotaur-0, 2015-08-16:
Hi
I'm reading the code of the accelerometre. but I didn't found where the noise is introduced. 
Is it simulated ?
https://github.com/morse-simulator/morse/blob/master/src/morse/sensors/accelerometer.py

adegroote, 2015-08-16:
In Morse design, noise are not implemented in sensors directly, but in a separate layer called modifiers. Depending what kind of noise you need, you can looks in Morse.modifiers if something feet your need. 

-------- Message d'origine --------

https://github.com/morse-simulator/morse/blob/master/src/morse/sensors/accelerometer.py

---
https://github.com/morse-simulator/morse/issues/649
Envoyé de mon téléphone Android avec K-9 Mail. Excusez la brièveté.

adegroote, 2015-08-26:
Can I close it ?

closed 2015-08-26 by minotaur-0