morse-simulator/morse #650: Morse 1.3 fails to build under Debian

sylvestre, 2015-08-17:
```
writing output... [  0%] components_library                                    
Exception occurred:
  File "/usr/lib/python3/dist-packages/sphinx/jinja2glue.py", line 159, in get_source
    raise TemplateNotFound(template)
jinja2.exceptions.TemplateNotFound: default/layout.html
The full traceback has been saved in /tmp/sphinx-err-cjpinjct.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
```

adegroote, 2015-08-21:
Can you provide the command line you use ? And the version of software you use ? The file is in the distribution and the conf.py does not change significantly since 1.2 so it looks like very strange.

sylvestre, 2015-08-24:
This was this bug:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=792711

commit 4126ec86 (2015-08-26): [doc] Make morse-theme layout also work with sphinx 1.3+

Fix #650
commit ded2fcbc (2015-08-26): [doc] Make morse-theme layout also work with sphinx 1.3+

Fix #650
closed 2015-08-26 by adegroote