morse-simulator/morse #651: Blender Scenes loaded incorrectly into morse

fdevillalobos, 2015-08-21:
I have been using morse for a while, and let me tell you, it rocks. Makes life so much easier to simulate and interface with real controllers.

While I get scenes more and more complicated, I found the need to introduce a new scene to my blender file, to use it as a background half SkyDome for one of my cameras. When I simulated, this inside of Blender, it works great, as only one scene is rendered and the other one can be safely used as a background.

But when I use the same file and load it through morse, I get all the object in the "Background" scene, copied in both the 512x512x scene, and the MORSE_LOGIC scene (By the way, can you briefly explain to me why there are two identical scenes when you run morse? Would be useful).

So, I have my Dome THREE times, in 512x512, in Morse_logic, and in Background scenes.
The problem is that it's now obviously rendered on my normal morse scene, besides being ALSO used as a background scene... a huge mess. haha.

So, is there any fix for this, does it have to be corrected? Where should I look to see if I can figure out a fix in morse code and submit a pull request.

Thanks a lot for all the support guys!
Francisco.

pierriko, 2015-08-21:
Hello,

Blender render resolution per scene, so for each different camera resolution, you need to have a different scene.

Our current workaround is to copy the environment for each resolution and apply every object position at every simulation tick.

You can find code in both files:
```
src/morse/sensors/camera.py:    def _update_scene(self):
src/morse/builder/environment.py:    def _cfg_camera_scene(self):
```

Cheers,
Pierrick

adegroote, 2015-08-21:
Can you provide the blend scene (with the background stuff) you try to load in Morse ?

fdevillalobos, 2015-08-21:
Hi, thanks for the quick response. I will try to make all my cameras the same resolution first and see if the 512x512 scene disappears, but I doubt that it will solve my problem. As requested, I attach here a link to the blend file that has two texture images packed in it.

Main Scene: FPS scene with just a floor and a rigid body camera that can walk around (WASD and look around with the mouse).

SkyDome_1 Scene: SkyDome and camera that can only move around with the mouse. Acts as a background scene for the camera in the main scene, maintaining the sky always in the exact same position giving the appearance of a far away sky without the need of having a far camera clipping very big.

When you run it inside blender, just using P in the Game Engine, it works perfect. You can see the background really far, and "play" the game.

When you launch it from Morse, this happens:
[image]
The dome gets rendered as an object, and also as a background.

I here also put the "code" for a simple default file in a minimal test environment:
```
from morse.builder import *

robot = Morsy()
robot.translate(2.0, 5.0, 0.0)
robot.rotate(0.0, 0.0, 3.5)
motion = MotionVW()
robot.append(motion)
keyboard = Keyboard()
robot.append(keyboard)
keyboard.properties(ControlType = 'Position')

env = Environment('/Users/prenav/Dropbox/DEV/Blender_vault/Environments/Background/SkyDome_2.blend', fastmode = False)
env.set_camera_location([-18.0, -6.7, 10.8])
env.set_camera_rotation([1.09, 0, -1.14])
env.set_camera_clip(clip_start=0.1, clip_end=3000)
env.set_camera_speed(speed=10.0)
```

Blender File:
https://drive.google.com/file/d/0B1RlgU5NtgbRV1BOT2NxMGNuR1k/view?usp=sharing

Thanks again for the help. It could be great if we could just define in mose the "Main" Scene name to load into MORSE_LOGIC, and have additional scene objects NOT copied over to MORSE_LOGIC so we can do stuff like this.

Thanks a lot,
Francisco.

adegroote, 2015-08-31:
I start to analyse the issue. As I understand the issue, the problem is in the method "append_meshes", which extracts all objects from a file, and insert in the main scene. In the Objects part of Blender file, all objects from all scene are mixed so it is a bit messy to use that in the case of multi-scene environment. 

One way may be to load file by scene, and copy object from "main" scene to the "default main scene of Morse", but not other objects. Moreover, we need a way to identify from the builder stuff the "main" scene from a Blender file.

adegroote, 2015-08-31:
Please check my branch fix_651. I still need to test it with a video_camera sensor.
You need to modify your builder script in the following way:

```
env = Environment('/Users/prenav/Dropbox/DEV/Blender_vault/Environments/Background/SkyDome_2.blend, main_scene = ...)
```

fdevillalobos, 2015-08-31:
Thanks A LOT for this. I don't want to leave you hanging! I'm busy today with something else, but I will try this as soon as I can and get back to you on the results! Thanks a lot!

fdevillalobos, 2015-09-03:
Hey Arnaud,

Thanks for the time to do this. I have just checked the branch.

1) Objects don't get imported any more to S.MORSE_LOGIC scene, which is great! First problem solved!

2) We have no way at least for what I've seen in the code of the environment creator to set the background_scene in morse. I digged a little bit into it and the solution should maybe be something like this:
(This is how I did it inside Blender): bpy.data.scenes[1].background_set = "scene_object", so in your case it would be something like:
bpymorse.get_context_scene().background_set = "scene_object", for what I have seen in the environment creator.
We could have some method like env.set_background_scene('scene_name').

3) All the same in the case where we just have a background for the camera, this works great with no additional code.

I hope some time soon I will be able to start doing pull requests and helping you with the code instead of just asking for the changes. Feels I'm getting there.

Once again, thanks a lot for the changes! They really work great! Thanks!

adegroote, 2015-09-03:
I added a method 'set_background_scene' in my branch which should do the job. Please confirm me it works as you expect it. I will merge them in this case.

fdevillalobos, 2015-09-03:
Thanks!
Works perfectly. We can close this issue now I guess. I will be using your branch until it gets incorporated to the official morse release. One tiny thing I've been having with your branch, every time I try to kill morse (ESC), it takes a long while and then I get:

Writing: /tmp/morse_default_autorun.crash.txt
Segmentation fault (core dumped)

Don't know what's different about it. Just leaving this here in case it means something to you.
Thanks again!

commit b3aab512 (2015-09-04): [builder] Try to handle in a more smart way environment with multiples scenes

Fix #651
closed 2015-09-03 by fdevillalobos