morse-simulator/morse #653: Kinect-based interface to control the human avatar

severin-lemaignan, 2015-08-28:
This issue tracks the development of an interface that would allow a human user to control a human avatar inside the simulation via a Kinect-like device.

This would enable a large range of HRI experiment to take place in the simulator.

Some related previous work:
- Some code, which relied on a ``pocolibs`` adapter for the Kinect (``niut``) [already exist](https://github.com/morse-simulator/morse/blob/master/src/morse/middleware/pocolibs/actuators/niut.py)
- This has been presented in a [paper](https://hal.archives-ouvertes.fr/docs/00/66/70/26/PDF/lemaignan2012morse.pdf).

maccradar, 2016-01-07:
What's the current state on these developments? Which code should I fork if I want to control the positions of the joints of the human avatar or can this be established via the place_IK_target and move_IK_target services?

severin-lemaignan, 2016-01-07:
@binnurr has been working recently on it, but it is not 100% clear to me what is the current status (cf https://github.com/binnurr/morse/tree/skeleton_tracking). @wrap1337, can you possibly update us?