morse-simulator/morse #674: Do not use `self.p...` as variable name to denote `previous values`

severin-lemaignan, 2015-10-29:
Several sensors (namely `velocity`, `gps`, `imu`, `attitude` and `accelerometer`) use the naming convention `self.pXXX` to denote a previous version of `XXX`. This leads to variable names like `self.patt`, `self.pp`, `self.plv`, `self.pq` that are maybe meaningful to the original author, but certainly not generally readable.

I would recommend to adopt instead `self.prev_XXX` which is explicit, PEP8 conformant and grep-able.

adegroote, 2015-10-29:
Do we really want to enforce pep008 for physics stuff (it will basically kill x,y,z, p,q,r, dx, dy, dz which have all a clear meaning) ?

severin-lemaignan, 2015-10-29:
Well, at the end of the day, the overall readability is what matters: `x`, `y`, `z`, `dx`, `dy`, `dz`, `t` are common and clear.

`p` (for position, I assume?), `q` (for quaternion?), `r` (for rotation?) are less convincing... IMHO `pos`, `rot` are short and clearer, I would prefer the full `quaternion` for `q`.

adegroote, 2015-10-29:
p,q,r together are body angular rates. q alone mean quaternion.

severin-lemaignan, 2015-10-29:
`p`, `q`, `r` are angular rates? hum I'm not familiar with these notations... using which rotation representation? yaw pitch roll?

Note that `p` alone may also mean position, like [here](https://github.com/morse-simulator/morse/blob/master/src/morse/sensors/velocity.py#L46)...

I do think it would be really useful to have non-ambiguous naming conventions!

adegroote, 2015-10-29:
> I do think it would be really useful to have non-ambiguous naming conventions!

Nobody can be against this idea.  Through, it is a little chimera. The same symbol is used in physics / math depending of the context. The same word can be used to denote different stuff between different  community (automaticien vs robotics) ... 

Moreover, considering the limited time of people working on the problem, can we really handle such "cosmetic" issues ?