morse-simulator/morse #677: Morse does not load with Blender 2.76.2

maccradar, 2015-11-11:
I've been using the morse-simulator ubuntu package with Blender 2.76.2 and it worked fine, but since it did not have the IMU socket patch for streaming the data to a socket, I am now trying 1.3_stable or master from git.

Both seem to hang when booted:

```sh
4:59:23 (1.3_STABLE) ~/workspace/morse/build$ morse check
* Checking up your environment...

* Running on Linux. Alright.

* Found MORSE libraries in '/usr/local/lib/python3/dist-packages/morse/blender'. Alright.

* Trying to figure out a prefix from the script location...

* Default scene found. The prefix seems ok. Using it.

* Setting $MORSE_ROOT environment variable to default prefix [/usr/local/]

* Checking version of /usr/local/bin/blender... Found v.2.76.2

* Version 2.76.2 of Blender is untested but should work

* Found Blender in your PATH
(/usr/local/bin/blender, v.2.76.2).
Alright, using it.

<hangs here>
```

Any ideas?

adegroote, 2015-11-11:
On which distribution do you have blender 2.76.2 ? Does it work properly with older version of Blender ?

adegroote, 2015-11-16:
Seems to work fine here, with the current HEAD. Can you tell more about your system ?

It seems to hang in the python version detection ? Which python version do you use ? Can you show the result of 

```
import sys
print('>>>' + '.'.join((str(x) for x in sys.version_info[:3])))
print('###%s' % sys.path)
```

in blender python console.

maccradar, 2015-12-07:
I've downgraded to 2.72 now which works fine. Only thing I needed 2.76.2 for was the blender Collada plugin but I am not using any Collada models anymore at the moment
for completeness here's the output of sys.version and sys.path:

>>>3.4.3
###['/home/jphilips/.config/blender/2.72/scripts/addons', '/usr/share/blender/scripts/addons', '/usr/share/blender/scripts/startup', '/usr/share/blender/scripts/modules', '/home/jphilips/workspace/czmq/bindings/python', '/home/jphilips/workspace/zyre/bindings/python', '/home/jphilips/research/acfr/simulator', '/home/jphilips/research/acfr/repos/filter', '/usr/lib/python3.4', '/usr/lib/python3.4/plat-x86_64-linux-gnu', '/usr/lib/python3.4/lib-dynload', '/usr/local/lib/python3.4/dist-packages', '/usr/lib/python3/dist-packages', '/usr/share/blender/scripts/freestyle/modules', '/usr/share/blender/scripts/addons/modules', '/home/jphilips/.config/blender/2.72/scripts/addons/modules']

As you can see it points to 2.72 now.

closed 2015-12-07 by maccradar