morse-simulator/morse #683: Real-time mode and fps > 60 does not work very well

adegroote, 2016-01-11:
When specifying 120 fps (for example, but it simplifies a lot the demonstration), an user expects to have sensor update every 8.33 ms, but it is not what happens at all. In fact, it gets:

- update
- ~1ms & update
- ~15ms & update
- ~ 1ms & update
- ~15ms & update

because the bge is mainly scheduled around the graphic update (which always runs at 60 fps). Basically, it does:
  - run all the physics / logic steps as fast as possible (but with the good "physical timestep")
  - run the display update
  - sleeps until the next iteration

A possible work-around is to put some sleep at the end or beginning of each logic step, through it is a bit difficult to compute the right duration of sleep (one need to recompute the internal blender logic to put some sleep only in the non-display update). Moreover, the sleep duration may vary in the time, depending computer load, so we may need an automatic adjustment policy

adegroote, 2016-01-14:
I start to investigate the issue more seriously and come to the conclusion there is no 'perfect' solution. The main limitation is the design of the BGE, and more precisely the fact that the graphic update is done in the same thread the logic / physics.

Solution 1
-------------
 
Sleeping in BestEffortStrategy.update(). It works not so bad, but computing automatically the "right" time to sleep is hard (as it depends of the number of logic step by graphical update, time elapsed in computation of the different phases, etc ...). I also tried some PID approach to auto-compute the sleep value, but we are always late over the change, and it works often worse. Moreover, the issue of such approach is that it basically assume that each frame takes almost the same time, but it is not true at all (because of the graphical update, and sensors / actuators with lower frequencies). Redoing the internal logic to compute when the graphical update will be done is almost impossible with the exposed data by BGE.

Solution 2
-------------

Another approach is to start a thread which sleeps 1/fps, and add a sync point (Event, semaphore, whatever you want) with the "main thread" The idea works, through it is not very reliable (due to the CPython implementation). When sleeping for 10ms, I often see the thread awaken after 15ms. I tried a signal approach too, but I'm not confident on its reliability (because Blender can catch the signal before us), and there is no Windows support.

Solution 3
-------------

Do nothing in Morse. Use an external syncer (and the socket time synchronization stuff). It is not perfect through it is actually the best solution found. As solution 2, it often implies a reduction of the speed of the graphic update, but it frequency output more reliable. Moreover, several experiments show that to have more precise output, it is better to put an higher global fps, and specify the frequency of the different sensor. For example, if you want some sensors at 100Hz, it is 'more' reliable (from a frequency point of view) to run the simulation at 300 fps, and specify explicitly a frequency of 100 for the sensor, than to run the simulation at 100 fps (it also may be related to the fact that 300 is a multiple of 60 :)).

I will continue to investigate possible solution, but if I don't find a better solution, I will write a simple 'morse_sync' tool which allow to synchronize the bge loop using the socket time synchronization mechanism, and  add some documentation in https://www.openrobots.org/morse/doc/latest/dev/time_event.html

severin-lemaignan, 2016-01-14:
At first sight (and without having looked at it beyond your bug reports!!), Solution 3 seems indeed the way to go. We need to make sure we start this 'morse _sync' tool automatically and transparently for the user, via a subprocess for instance.

severin-lemaignan, 2016-01-19:
[Commenting here about #689 as it is a high-level discussion, better tracked by this issue] One 'high-level' comment: the so-called 'internal syncer' basically prevent other external sources of time synchronisation (like what would be provided by HLA, rostime, etc) to be used. It may or may not be want we want. But in any case, we need to clarify globally the management of time in MORSE.

As it stands today, here is a list of options on the environment that impact the time behaviour (copy-paste from [1]):
- `set_speed(fps=0, logic_step_max=0, physics_step_max=0)`
    Tune the speed of the simulation
-  `set_physics_step_sub(step_sub)`
    If you increase the step_sub value, your simulation will spent more time in computing physics, but it will be more precise. The default value is 2.
-  `set_time_strategy(strategy)`
    Choose the time strategy for the current simulation  [btw, this doc is not very useful!!]
- `set_time_scale(slowdown_by, accelerate_by)`
- and now: `use_internal_syncer()`

Besides those, we can setup fps at a component level with the `component.frequency()` method.

...I'll bet people (including devs!) are going to be lost!

So, what about consolidating everything in a single method, called `time`. Something like:
```python
class builder.Environment:
    #....
    def time(factor = 1.0,
                  frequency = 60,
                  external = False,
                  allow_skipping_frames = False,
                  physics_accuracy = 2)
    ''' Configures how MORSE manages time. Use this method to accelerate (faster-than-real-time
    simulation) or slow-down the simulation, synchronize it with external time sources, set the update
    frequency of sensors, configure the balance accuracy/speed, etc.

    Time is a complex matter, with several factors impacting the actual accuracy of the simulation: how
    complex is the scene, how powerful is your simulation machine, etc. If MORSE can not reach the 
    update frequencies that you have configure with this method, it will try its best to give you useful
    warnings. 

    :param float factor: the 'time factor' (> 0.0, default: 1.0). Values above 1.0 accelerate the simulation
    (factor=2.0 means that the simulation will run twice as fast as real-time: 1 real second correspond 
    to 2 seconds in the simulated world); values between 0 and 1.0 slow down the simulation. 
    :param float frequency: the main update frequency of MORSE. This sets the default frequency (in
    hertz, default: 60Hz) at which all the sensors and actuators are updated. Note that you can override
    this frequency for each individual component by using component.frequency(...).
    :param bool external: if True, MORSE will listen on a socket for time steps. Useful when your
     system provide a time synchronisation mechanism. See [...] for details.
    :param bool allow_skipping_frames: if set to True (default: False), MORSE may skip simulation
    frames if necessary to keep up with the requested simulation speed.
    :param integer physical_accuracy: the number of physical simulation substeps performed by the 
    Bullet physics engine. The higher, the more accurate the physics, but it may impact the frequency of 
    the simulation main loop. The default value (2) should be fine in most cases. 
```
Then, if `external` is not set, we use (transparently) `morse_sync` to ensure correct time synchronization.
I'm not sure about 'time strategies': do they need to remain? shoud `Fixed simulation step` become the new default? I'm not very clear about this point...

Of course, this `time()` method is a draft, for discussion. This design can certainly be improved.

[1] http://www.openrobots.org/morse/doc/latest/user/code/morse.builder.html#module-morse.builder.environment

adegroote, 2016-01-20:
>  One 'high-level' comment: the so-called 'internal syncer' basically prevent other external sources of time synchronisation (like what would be provided by HLA, rostime, etc) to be used. It may or may not be want we want. But in any case, we need to clarify globally the management of time in MORSE.

Yes. It is why I provide it as an option which must be explicitly configured BTW, at the moment, there is no way to make Morse follow a ROS-clock (through we can emit a ROS-clock with Morse).

Considering the time method, I don't see how it improves the situation, you just make all the argument in one function, but you still need to configure the same amount of stuff. And it breaks compatibility. 

Moreover, it has no chance to work at all with the current implementation.  As explicated in several place,  ``bpymorse.set_speed`` must be called before the creation of any components, so you can't include in ``env.time()``. (The reason is that the frequency() is computed on the flight, on the base of the current frequency of the simulator). 

Concerning  ``set_physics_step_sub``, it is not related to time-management, but to physics accuracy (of course, it implies a computation time penalty, but not worse than adding a camera).

I think ``allow_skipping_frames`` is not a good name, it is how it is implemented in Blender, but not a good information of semantic. ``BestEffort`` and ``FixedTimeStep`` looks better.

> Then, if external is not set, we use (transparently) morse_sync to ensure correct time synchronization.

I don't think it is a good idea. If you use the default setup (i.e. BestEffort at 60Hz) (that most people use I guess), morse_sync does not help (at contrary). It is only needed if you have high-frequency simulation (and the higher-frequency, the best). 

 > Shoud Fixed simulation step become the new default? 

Fixed should not be the default, as (I think, I don't have stat) most of people don't have clock-synchronization between Morse and their functional layer.

severin-lemaignan, 2016-01-20:
> Considering the time method, I don't see how it improves the situation, you just make all the argument in one function, but you still need to configure the same amount of stuff. And it breaks compatibility.

It does improve the situation by grouping all the parameter in one location: having +/- 5 different methods to configure time management is not a good user interface design, and will also certainly lead to a frustrating experience for user who will forget to setup something (if only because the role of 'time_strategy' for instance may not be exactly obvious).

We need to design our interface from the *sole* perspective of the user. What do the users want, related to 'time' in a broad sense? My guess is (by order of importance):
- ensure the simulation 'is not slow', ie 'good performances'. And if the scene is too complex for their system, they may want that the simulator gives them a useful indication (something like 'I'm not capable to keep up with the requested speed: I'm drifting by 0.01 seconds at every step! Consider simplifying your simulation or enable 'allow_frames_skipping'' -- exact message to be discussed)
- faster-than-real-time simulation
- (for some more advanced users) the frequency at which sensors/actuators operate
- (for some more advanced users) external time synchronisation

IMHO, everything else (like the 'fps') is internal and should not generally be used. Up to MORSE to adopt the best strategy to fullfill the above objectives.

My proposal of a single `time` method can possibly be implement as a high-level wrapper for the various other methods (so that we keep the other methods for backward compatibility), but I do think having a single entry-point for time related configuration would improve the situation. For me at least :-)

> Moreover, it has no chance to work at all with the current implementation. As explicated in several place, bpymorse.set_speed must be called before the creation of any components, so you can't [...]

In term of user interface, I would call it a bug. This should be fixed if possible. I don't know how feasible it is, but I do know that having *one special* environment configuration command that need to be used before anything else is confusing and error prone.

> 
> Concerning set_physics_step_sub, it is not related to time-management, but to physics accuracy (of course, it implies a computation time penalty, but not worse than adding a camera).

The point I was making is: to someone who is investigating why his simulation runs slowly, it is one more 'time related' tuning option that adds to the confusion.

> 
> I think allow_skipping_frames is not a good name, it is how it is implemented in Blender, but not a good
> 
> information of semantic. BestEffort and FixedTimeStep looks better.

Honestly, out of context, I would expect that MORSE's 'best effort' is to ensure a 'fixed time step': I'm not kidding here. The name 'BestEffort' and 'FixedTimeStep' may be very clear to you, but I suspect they are not clear to most users (RTFM does help a little, but when people are encountering slow-downs in their simulations, they may try one or the other strategy semi-blindly, without fully understanding what it means for the simulator). IMHO, `allow_skipping_frames` is a *bit* better (but still arguably not very good).

> Then, if external is not set, we use (transparently) morse_sync to ensure correct time synchronization.
>
> I don't think it is a good idea. If you use the default setup (i.e. BestEffort at 60Hz) (that [...]

Well, that's the point: if we always start morse_sync when there is no other sources of external time synchronization, then we should be able to use FixedTimeStep, shouldn't we?

adegroote, 2016-01-20:
> We need to design our interface from the sole perspective of the user. What do the users want, related to 'time' in a broad sense? My guess is (by order of importance):

We are not user, at least not basic one, so our guess is not evidence, but no need to bikesheed on the question.

 > faster-than-real-time simulation

I'm sure lot of people want that. And I'm also sure that most of them don't have the software architecture to make it really works (i.e. sync properly your functional clock to the simulated one, and have algorithm performant enough to be really executed more often).

> the frequency at which sensors/actuators operate

that is strange, I would expect to be number 1 need to have a realistic simulation. Having an information at 10Hz instead of 100Hz has large issue on your real system and your algorithms. The simulator is faster enough if it can provide the data at the expected frequency, that is the only useful metric (well, the viewport update can be a metric, but not so important in the end).

> In term of user interface, I would call it a bug. This should be fixed if possible. I don't know how feasible it is, but I do know that having one special environment configuration command that need to be used before anything else is confusing and error prone.

I tend to agree, but never have time to work on it. The solution is not too hard I think:
  - store the frequency at the call level
  - in environment.create, parse all the objects created, and configure the delay properly
At a side effect, it can allow to choose  automatically a correct fps, allowing to approach better the requested frequencies.

> The point I was making is: to someone who is investigating why his simulation runs slowly, it is one more 'time related' tuning option that adds to the confusion.

To be honest, I never saw a Morse simulation which physics is the issue.  And you can have a clear information of that in the framerate information (% of use of physics). 

> Well, that's the point: if we always start morse_sync when there is no other sources of external time synchronization, then we should be able to use FixedTimeStep, shouldn't we?

No, it won't work at all if you request  for a 200Hz simulation. In FixedTimeStep (i.e allow_skipping_frames = False), you always have:
  - call logic
  - call physics
  - call graphic update
If you can ensure the full loop in less than 1/200 sec, it will work. If you cannot, it won't work (and morse_sync won't help). To run at 60Hz, it will generally work, but as good as 'BestEffort' mode, even without morse_sync.

To "almost guarantee" fixed logic step for fast simulation and "real-time", the best thing to use is "BestEffort" with a big value of max_logic_step / max_physics_step (by big, I mean 20 for example, the Blender documentation says that 5 is the maximum, but it is a lie) and morse_sync.

Definitively, time is a hard beast, particularly due to the BGE implementation :)

severin-lemaignan, 2016-01-20:
> Definitively, time is a hard beast, particularly due to the BGE implementation :)

Agreed :-)

adegroote, 2016-01-21:
An idea to improve the situation without changing any kind of code would be to provide some documentation guidelines. I think about 4 scenarios:

Scenario 1 : basic need
------------------------------

Do nothing

Scenario 2 : fully reproductive simulation // synchronisation with other simulators
------------------------------------------------------------------------------------------------------

- FixedTimeStep 
- env.configure_manager(['hla'|'socket'], sync_time = True) 
morse_sync is not a good solution here

Scenario 3: simulation with high-frequency (> 60) sensors | actuators
---------------------------------------------------------------------------------------

- bpymorse.set_speed(large_fps, 20, 20)
- BestEffort
- specify frequency of your sensors/actuators
- use_internal_syncer

Scenario 4: faster_than_realtime
------------------------------------------

Do the same than scenario 3

severin-lemaignan, 2016-01-21:
This sounds like a good idea for now. I still think there is room for improvement in the Builder API for time management. But as a first step, describing the 4 scenarios you suggest sounds like a good idea.

I would even add a note at the end of this documentation page: If you encounter other kind of time issues/requirements, please please send a mail to morse-users so that we better now what is needed!

adegroote, 2016-01-22:
turning v_sync to OFF completely change the problem. Need to rethink about it. But why it is not the default in Morse ? :D

adegroote, 2016-02-03:
Definitely closed by 75a8fb0. Github is not smart enough :D

closed 2016-02-03 by adegroote