morse-simulator/morse #701: Cannot build version 1.4

andystevens91, 2016-03-11:
Hi!
I just started using MORSE today. I'm trying to get a working installation of MORSE 1.4 in order to use [morseweb](https://github.com/yarox/morseweb), but, no matter how I install it, I always get v1.2.
I tried:

1. installing using `apt-get install morse-simulator` --> `morse -v` 1.2
2. building from git branch master  --> `morse -v` 1.2
3. building from git branch 1.4_STABLE  --> `morse -v` 1.2
4. downloading the tarball of 1.4 and building from there  --> `morse -v` 1.2

After every try I've nuked the installation and tried a different method, but `morse -v` is ALWAYS 1.2.
What am I doing wrong?

adegroote, 2016-03-12:
On 11/Mar - 13:05, Andrea Di Stefano wrote:
> Hi!
> I just started using MORSE today. I'm trying to get a working installation of MORSE 1.4 in order to use [morseweb](https://github.com/yarox/morseweb), but, no matter how I install it, I always [...]
Do you remove the morse-simulation debian package ? I guess you still
get the Morse version installed in /usr/bin from the debian package.

andystevens91, 2016-03-12:
Yep, with `apt-get --purge remove morse-simulator`

adegroote, 2016-03-14:
In which prefix do you install Morse? what returns the command 

which morse

andystevens91, 2016-03-14:
`/usr/local/bin/morse`, which I think is the default prefix for the cmake installation.

pierriko, 2016-03-21:
you can check if `${PREFIX}/lib/python3*/*-packages/morse/version.py` is correct

andystevens91, 2016-03-28:
Sorry, I have been busy this week and did not check for new replies.
In the end I did it by building from the 1.4 tarball AFTER formatting the disk and reinstalling the OS, so I suppose there was still something left from the previous installations.
Thanks for your help everyone!

closed 2016-03-28 by andystevens91