morse-simulator/morse #702: Changing VideoCamera field of view

thushv89, 2016-03-17:
Hi, 
I would like to know if there is a way to change the field of view of the video camera. I want an wide-angle image of my robot's view.

I tried tweaking cam_width and cam_height but had no luck!

My script for camera looks like this
```
camera = VideoCamera()
camera.translate(x = 0.5, z= 0.6)
camera.rotate(y = -0.13)
camera.properties(cam_width=1024,cam_height=256,cam_far=500)
camera.frequency(5)
atrv.append(camera)
camera.add_interface('ros',topic='/camera')
```

MORSE version: 1.3
Blender version: 2.69.0
Python version: 3.4.3

Thank you

thushv89, 2016-03-17:
So after playing with this issue for a while, I found a hack. So, if you move the camera to the back of your robot and set cam_near high enough so your view is not obstructed by the robot, for example,
```
camera.translate(x = -1.5)
camera.properties(cam_near=1.5)
```
then you can simulate the wide-angle effect. 

But I'm not sure, if there's a more clever legitimate way of doing this. So I'm leaving the issue open.

adegroote, 2016-03-21:
The  KX_Camera seems to have a .fov member, so I think you can change the field of view using this field. Unfortunately, for the moment, there is no interface for that in Morse.

pierriko, 2016-03-21:
try changing the [focal length](https://www.blender.org/api/blender_python_api_2_72_release/bpy.types.Camera.html#bpy.types.Camera.lens) with:
```python
camera.properties(cam_focal = 1)
```

flixr, 2016-03-21:
To convert the cam_focal parameter to opening angle: `arctan(cam_focal / 16)`

thushv89, 2016-03-22:
Yup, reducing ` cam_focal` did the trick. Thanks everyone. Really appreciate it. :) Closing the issue

closed 2016-03-22 by thushv89