morse-simulator/morse #703: camera/image topic not displaying the image properly

thushv89, 2016-03-22:
Hi I am encountering a somewhat strange issue with /camera/image topic. I'm try to see what robot is seeing through its camera (using web_video_server package). And the image at current location is not properly rendered In other words, only a portion of the /camera/image appears to be the current view of the robot (this part gets updated), but the rest of the image is something robot saw previously (this part is kept constant). Refer the image below.

[image]

I'm currenlty at the following location
[image]

My code looks like this (I also have an Odometry, Sick and Hokuyo laser scans)
```
atrv = ATRV()
atrv.translate(x=2.5, y=3.2, z=0.0)

frequency = 3
camera = VideoCamera()
#camera.translate(x = -1.5, z= 0.9)
camera.translate(x = 0.7, z= 0.5)
camera.rotate(y = -0.0)
#camera.properties(cam_width=256,cam_height=192,cam_far=500,cam_near=2.15)
camera.properties(cam_width=256,cam_height=192,cam_focal=10.)
camera.frequency(frequency)
atrv.append(camera)
camera.add_interface('ros',topic='/camera')
...

```

Reproducing the Error
-----------------------------
I could not infer an exact reason for this strange behaviour. But commonly, it seems to occur when the robot stays at one place for a long time (15-30 mins) and then start moving. However, in some instances, it could happen as soon as morse started. 

Any idea why this is happening?

MORSE version: 1.3
Blender version: 2.69.0
Python version: 3.4.3

adegroote, 2016-03-22:
It may  be related to #564. I'm wondering if it is an issue in core Morse, or if it is related to ROS bindings (as it seems to only happen in ROS case). Can you, by any luck, try with yarp / yarpview ?

thushv89, 2016-03-22:
@adegroote I watched the videos given in the link. Though both these cases are related to issues of the video camera, it seems nature of the issues is quite different in two cases. 

It seems the issue surfaces if the robot stay stationary long after starting morse. If I move the robot around in the beginning for some time, it works fine. 

I'll have a look at yarp and let you know!

thushv89, 2016-03-31:
So I was able to reproduce the error the following way.
1. Start ROS
2. Start Morse
3. Start web_video_server

Now resize the size of the morse simulator window (Specifically make it smaller). And when it's small enough, you will see the same effect elucidated in this issue coming to play.

It seems it is possible to get around the error by making the window a bit bigger.

adegroote, 2016-04-01:
It is a known issue with Blender rendering. There is a partial workaround in 7db36ec4273a098edaf240094f274b0f5fa2694b, but we only consider the window size at initialization. If you change the size of the windows during runtime, bad thing can definitively happens.

The real fix is still pending in  Blender (i.e use FBO, see https://developer.blender.org/D1618).

adegroote, 2016-04-01:
To precise the issue, the current Blender rendering techniques is related to the viewport, and so the size of the "Blender windows". You cannot have a sensor with a resolution bigger than the one of the main Blender. So, for now, just get a big Blender and move it in a non-used virtual desktop (good wm can do it automatically :))