morse-simulator/morse #704: Installing Morse problem in Macbook Pro

jim1949, 2016-04-04:
*Thanks for taking the time to fill an issue with MORSE!*

[Describe here your issue]When I tried to install the morse following the instructions, there's some problems shown in my command line.

## Paste here a (minimal) Builder script that triggers the issue

jjdeMacBook-Pro:bin jj$ brew install morse-simulator
==> Installing morse-simulator from morse-simulator/morse
==> Installing dependencies for morse-simulator/morse/morse-simulator: cm
==> Installing morse-simulator/morse/morse-simulator dependency: cmake
==> Downloading https://homebrew.bintray.com/bottles/cmake-3.5.1.el_capitan.bott
######################################################################## 100.0%
==> Pouring cmake-3.5.1.el_capitan.bottle.tar.gz
==> Caveats
Emacs Lisp files have been installed to:
  /usr/local/share/emacs/site-lisp/cmake
==> Summary
🍺  /usr/local/Cellar/cmake/3.5.1: 2,010 files, 27.5M
==> Installing morse-simulator/morse/morse-simulator
==> Cloning https://github.com/morse-simulator/morse.git
Updating /Library/Caches/Homebrew/morse-simulator--git
==> Checking out tag 1.4
==> cmake . -DCMAKE_INSTALL_PREFIX='/usr/local/Cellar/morse-simulator/1.4' -DCMA
Last 15 lines from /Users/jj/Library/Logs/Homebrew/morse-simulator/01.cmake:
-- Check for working CXX compiler: /usr/local/Library/ENV/4.3/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
**CMake Error at /usr/local/Cellar/cmake/3.5.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.5.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.5.1/share/cmake/Modules/FindPkgConfig.cmake:48 (find_package_handle_standard_args)
  CMakeLists.txt:27 (find_package)
**

-- Configuring incomplete, errors occurred!
See also "/tmp/morse-simulator20160404-3600-16mb0td/CMakeFiles/CMakeOutput.log".

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/morse-simulator/homebrew-morse/issues
# if your issue is not related to a specfic middleware,
# you can use the socket interface to ease the testing.
robot.add_default_interface('socket')

**what is the problem of this? And what should I do to solve this?**

adegroote, 2016-04-05:
As stated in the error message, you need to install pkg-config. 

```
brew install pkg-config
```

I will add in the brew recipe when I can

jim1949, 2016-05-05:
Thank you very much. This works now.

closed 2016-05-09 by pierriko