morse-simulator/morse #707: About Morse cannot recognize the Blender.

jim1949, 2016-05-06:
*Thanks for taking the time to fill an issue with MORSE!*

 I am installing the morse today again. And I am getting stuck in the step when I check the morse. It shows it cannot find the excutable blender. 
It shows like below:
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
eduroam-164-210:~ jj$ morse check
* Checking up your environment...

* MORSE has only been tested on Linux. It may work
on other operating systems as well, but without any guarantee

* Found MORSE libraries in '/usr/local/Cellar/morse-simulator/1.4/lib/python3.4/site-packages/morse/blender'. Alright.

* Trying to figure out a prefix from the script location...

* Default scene found. The prefix seems ok. Using it.

* Setting $MORSE_ROOT environment variable to default prefix [/usr/local/]

* Could not find a correct Blender executable, neither in the path or in MORSE
prefix. Blender >= 2.65 is required to run MORSE.
You can alternatively set the $MORSE_BLENDER environment variable to point to
a specific Blender executable

* Could not find Blender executable

* Your environment is not correctly setup to run MORSE!
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
My blender version is 2.76b, and I have found a question solution to this:
echo 'export MORSE_BLENDER={path_to_blender_executable}/blender' >> ~/.bashrc
So I added two commands:
echo 'export MORSE_BLENDER=/Applications/blender.app/Contents/MacOS/blender' >> ~/.bashrc

And 
MORSE_BLENDER=/Applications/blender.app/Contents/MacOS/blender
However, I still got the same result:
My Blender still cannot be found by the Morse.

Any suggestions?

- *MORSE version:* 1.4
- *Blender version:*2.76b
- *Python version:* 3.4.2

pierriko, 2016-05-09:
are you using bash ? to check if the variable is set up correctly, you can type the following in a terminal:
```
➤ echo $MORSE_BLENDER 
/path/to/blender
➤ $MORSE_BLENDER --version
Blender 2.72 (sub 0)
	build date: 2014-10-21
	build time: 16:50:57
	build commit date: 2014-10-03
	build commit time: 13:58
	build hash: 95182d1
...
```

pierriko, 2016-05-09:
Can you try this commit: https://github.com/pierriko/morse/commit/5630816b5c7c6adb186fefed2f351fa8ba33fb7c

If it works for you, I'll push to the main repository.
Thanks

gsubramani, 2016-06-29:
I'm getting the same issue with MAC OS X, python 3.5.1 and blender 2.77.

```
Could not find a correct Blender executable, neither in the path or in MORSE
prefix. Blender >= 2.65 is required to run MORSE.
You can alternatively set the $MORSE_BLENDER environment variable to point to
a specific Blender executable
```
was this tested?

pierriko, 2016-06-29:
Hi @gsubramani can you try the following patch:
https://github.com/pierriko/morse/commit/e31b4df71527588466eee290b205bbc57e906f7e

Let me know if it fix your issue and I'll push upstream

gsubramani, 2016-06-29:
@pierriko 👍  That seems to fix the problem!

commit 2573bfeb (2016-06-30): [bin] try to find blender on MacOS

it seems that blender is not in `which` on Mac OS.
[couldnt test myself]
fix #707 thanks to @gsubramani for testing.
closed 2016-06-30 by pierriko