morse-simulator/morse #708: Creating robots in a loop without explicit naming them makes their components disapear

nicolaje, 2016-05-18:
In the builder script, creating robots in a loop without explicitly naming them creates erratic robot naming and their attached sensors/actuators disappear.

Minimal builder file to reproduce the issue:

```python
from morse.builder import *

for i in range(1,10):
    robot = Pioneer3DX()

    differential_actuator=MotionVWDiff()
    differential_actuator.add_interface('socket')
    robot.append(differential_actuator)
    robot.translate(x=i)
    robot.add_default_interface('socket')

env = Environment('sandbox', fastmode = False)
```

Output of the simulation initialization:

```bash
...
[    0.200] Robots in the simulation:
[    0.200] 	ROBOT: 'Pioneer.005'
[    0.200] 		- Component: 'Empty.005'
[    0.200] 	ROBOT: 'Pioneer.007'
[    0.200] 		- Component: 'Empty.007'
[    0.200] 	ROBOT: 'Pioneer.006'
[    0.201] 		- Component: 'Empty.006'
[    0.201] 	ROBOT: 'Pioneer'
[    0.201] 		- Component: 'Empty'
[    0.201] 	ROBOT: 'Pioneer.001'
[    0.201] 		- Component: 'Empty.001'
[    0.201] 	ROBOT: 'Pioneer.002'
[    0.201] 		- Component: 'Empty.002'
[    0.201] 	ROBOT: 'Pioneer.003'
[    0.201] 		- Component: 'Empty.003'
[    0.202] 	ROBOT: 'robot'
[    0.202] 		- Component: 'robot.differential_actuator'
[    0.202] 	ROBOT: 'Pioneer.004'
[    0.202] 		- Component: 'Empty.004'
...
```

While when explicitly naming the robots in the builder script, the scene is correctly initialized and the robots naming conflicts is correctly handled

```python
from morse.builder import *

for i in range(1,10):
    robot = Pioneer3DX()

    differential_actuator=MotionVWDiff('robot') # Robot is named
    differential_actuator.add_interface('socket')
    robot.append(differential_actuator)
    robot.translate(x=i)
    robot.add_default_interface('socket')

env = Environment('sandbox', fastmode = False)
```

output:

```bash
...
[    0.251] Robots in the simulation:
[    0.251] 	ROBOT: 'robot_005'
[    0.251] 		- Component: 'robot_005.differential_actuator'
[    0.251] 	ROBOT: 'robot_007'
[    0.251] 		- Component: 'robot_007.differential_actuator'
[    0.251] 	ROBOT: 'robot_006'
[    0.252] 		- Component: 'robot_006.differential_actuator'
[    0.252] 	ROBOT: 'robot'
[    0.252] 		- Component: 'robot.differential_actuator'
[    0.252] 	ROBOT: 'robot_001'
[    0.252] 		- Component: 'robot_001.differential_actuator'
[    0.252] 	ROBOT: 'robot_002'
[    0.252] 		- Component: 'robot_002.differential_actuator'
[    0.252] 	ROBOT: 'robot_003'
[    0.252] 		- Component: 'robot_003.differential_actuator'
[    0.252] 	ROBOT: 'robot_008'
[    0.252] 		- Component: 'robot_008.differential_actuator'
[    0.253] 	ROBOT: 'robot_004'
[    0.253] 		- Component: 'robot_004.differential_actuator'
```

By the way, pymorse really needs an easy way to access robots from their name as a string, in the spirit of:

```python
...
with pymorse.Morse() as s:
    robot1_handle=s.get_robot(s.robots[0])
    robot2_handle=s.get_robot('my_robot')
```

Best regards,

Jeremy

---
- *MORSE version:* [morse 1.3-68-g5538]
- *Blender version:* [2.75.4]
- *Python version:* [3.4.2]

adegroote, 2016-05-18:
I don't want to be rude, but there is a red warning box in the [builder documentation](https://www.openrobots.org/morse/doc/stable/user/builder.html#adding-a-robot) about the creation of robot in a loop, explaining how it can be done properly. 

For the second point, if I understand properly, it has been fixed in 1.4, see #358.

nicolaje, 2016-05-19:
My bad, I think I remember having initialized robots in a loop without this precaution before, so I did not dig in the doc, should have seen the big red box.

Is there a reason why the close_context is not called by default in each robot constructor?

adegroote, 2016-05-19:
``close_context`` must be called at the end of the context (here the loop), it takes care of renaming and the proper link between the different objects. So I can't be called in the robot ctor (well, you can, but it will have no effect at all).

nicolaje, 2016-05-20:
Thanks for these clarifications. It seems that with 1.4 it causes a problem with wheeled robot losing their wheels though. I will create a dedicated issue if this is confirmed.

closed 2016-05-20 by nicolaje