morse-simulator/morse #710: Morse won't run with Blender 2.77

miscott, 2016-06-23:
I have been using Blender 2.69 with MORSE 1.4 (and ROS as middleware) on Ubuntu 14.04. It was working great. I wanted to upgrade to Blender to 2.77 as it is easier to speed up MORSE simulations, which will be beneficial for my research. 
Upon upgrading to Blender 2.77, MORSE will no longer run. The error message received when I run morse check on terminal is as follows:
* Running on Linux. Alright.

* Found MORSE libraries in '/usr/local/lib/python3/dist-packages/morse/blender'. Alright.

* Trying to figure out a prefix from the script location...

* Default scene found. The prefix seems ok. Using it.

* Setting $MORSE_ROOT environment variable to default prefix [/usr/local/]

* Checking version of /usr/bin/blender... Found v.2.77.0

* Version 2.77.0 of Blender is untested but should work

* Found Blender in your PATH
(/usr/bin/blender, v.2.77.0).
Alright, using it.

* Checking version of Python within Blender /usr/bin/blender... Found v.3.5.0

* Blender is compiled for Python 3.5.0 but MORSE has been compiled for Python 3.4.3! Check your MORSE build configuration or the selected Blender version.

* Bad Python version

* Your environment is not correctly setup to run MORSE!

Ideally, I would like to get MORSE and ROS to run with Blender 2.77, but if that is not possible I would at least like to go back to Blender 2.69 to run some more tests. 
Thanks in advance!

pierriko, 2016-06-23:
Hi, you can try to:
```
export MORSE_SILENT_PYTHON_CHECK=1
```
before running morse, this will disable the crash if Python missmatch, but it might lead to various errors, due to the way both your system and Blender python libs will interact (or conflict).

You can always get another blender version at http://download.blender.org/release/

miscott, 2016-06-23:
Thanks a bunch for that command. When I run it and then run morse check I get the reply:
"Your environment is correctly setup to run MORSE" as expected. The problem, as you mentioned, is more errors occurred. The errors are namely attribute and assertion errors.

pierriko, 2016-06-23:
Then you need to build Blender from source

miscott, 2016-06-23:
I'll try that, thanks again!

pierriko, 2016-09-30:
in case of future issue like this one, as stated in #715 it's now possible to run morse with various blender versions from d.b.o, you just have to setup the to environment variables:

```bash
export MORSE_SILENT_PYTHON_CHECK=1
export MORSE_POP_PYTHONLIB="/usr/lib/python3.5"
```

(assuming your system's Python3 is `3.5`).

closed 2016-09-30 by pierriko