morse-simulator/morse #711: Increasing simulation speed with ROS (Gmapping)

miscott, 2016-07-21:
I am currently running MORSE with ROS middleware. I'm running MORSE 1.4-33-g2573b with blender 2.77a and ROS indigo on Ubuntu 14.04. I am using blender 2.77 because of the abilities to increase simulation speed (my simulations can be quite long).
I am currently having trouble speeding up the simulation while getting MORSE to communicate with ROS. I use env.set_time_scale(accelerate_by=1.5) right now, and MORSE/Blender will launch fine. When I try to run my roslaunch files, however, the gmapping package will not work. No error messages are really given, it just won't run. It will, however, run great when there is no increase in simulation speed. 
My thoughts are that there is something wrong with time syncing between MORSE and ROS. I tried using env.set_time_strategy() for both BestEffort and FixedSimulationStep, but those do not solve the problem.

So my question simplified: Has there been a tested method for running the increased simulation time with ROS middleware and making sure that gmapping was communicating properly? Thanks! 

The MORSE environment I'm trying to run:
```
from morse.builder import *
from math import pi

robot = QUAD2012() 
robot.add_interface('ros')
robot.translate(x=0.0, y=0.0, z=0.5)
robot.rotate(z=pi/3)

# An odometry sensor to get odometry information
odometry = Odometry()
robot.append(odometry)
odometry.add_interface('ros', topic="/odom")

# Keyboard control
keyboard = Keyboard()
robot.append(keyboard)
keyboard.properties(Speed=3.0)

#QUAD2012 Motion control
motion = MotionXYW()
motion.properties(ControlType = 'Position')
robot.append(motion)
motion.add_interface('ros', topic='/cmd_vel')

#Lidar Specs
scan = Hokuyo()
robot.append(scan)
scan.properties(Visible_arc = True)
scan.properties(laser_range = 30)
scan.properties(resolution = 1.0)
scan.properties(scan_window = 270.0)
scan.create_laser_arc()
scan.add_interface('ros', topic='/base_scan')

# Set the environment
#env = Environment('indoors-1/indoor-1')
env = Environment('indoors-1/boxes')
#env = Environment('laas/grande_salle')
#env = Environment('tum_kitchen/tum_kitchen')
#env = Environment('outdoors')
env.set_camera_rotation([1.0470, 0, 0.7854])
env.set_time_scale(accelerate_by=1.5)
#env.use_internal_syncer()
env.set_time_strategy(TimeStrategies.BestEffort)
```

adegroote, 2016-07-22:
Short anwser: I never use time-acceleration with ROS so I don't have too much idea. Anyway, a first thing which shocked me is that you do not export the system clock using the clock sensor, so there is no way you can run at "simulated time".

After that, I would test individual topics using manually rostopic. If the behaviour of the simulator is right, I don't think there is specific issue on Morse side.

miscott, 2016-07-22:
Thanks for your comment. 
I'm a bit confused on the clock part. I need to export the time using Clock(), and that helps the simulation run at a proper time interval? Without exporting clock, the simulation is not necessarily timed properly? Is that correct? Thanks again.

flixr, 2016-07-22:
Yes, see http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2BAC8-clock_Topic

miscott, 2016-07-22:
Thanks so much!

closed 2016-10-05 by pierriko