morse-simulator/morse #743: Can not build Morse 1.4

abbasabibi, 2016-10-23:
Hello,

I'm trying to build morse 1.4, but it always installs the morse 1.2!! I also checked the version.py and thats what I get :

VERSION="1.4-1-g82c4c-dirty"
PROJECT_STABLE=True

I need the function env.set_time_scale which has been added in morse 1.4 . However
 morse -v ---> mordse 1.2 .

Do you have any fix?

adegroote, 2016-10-24:
How do you install morse 1.4 ? I suspect you have another install that is hiding the new one.

pierriko, 2016-10-26:
Can you give the output of the following:
```
python3 -c'import morse.version;print((morse.version, morse.version.VERSION))'
```

abbasabibi, 2016-10-26:
Hello,

Thank you for your suggestions. It turns out after removing the old morse i needed to run an
"apt-get auto-remove" and everything is fine now.

Best,
Abbas

closed 2016-10-26 by severin-lemaignan