morse-simulator/morse #752: MOOS new python interface

msis, 2016-12-16:
What is the best practice to update a middleware?

`MOOS` has now an official interface for [python](https://github.com/themoos/python-moos) and I want to contribute with a new interface to MORSE, but afraid if I replace the existing one by @davidhodo through a PR, it might ruin legacy code.

Should we have a `MOOS2` (or even `MOOSv10` as the creator likes to call it) interface in this case?

severin-lemaignan, 2016-12-16:
@msis Great to know that MOOS is still used!
What kind of breakages do you expect? would this change fundamentally how MORSE interact with MOOS?

I let  @davidhodo comment, but if the current user base for MOOS is small (as I suspect), then I would favour just replacing the current implementation with the official `moos-python` bindings. Carrying over legacy code is not really my cup of tea unless we have solid reasons to do so.

msis, 2016-12-16:
In `MOOSv10`, asynchronous communication with the `MOOSDB` which allows having callbacks for different messages in addition to the traditional way of getting mail implemented here.

I will be replacing the docs too all along.

davidhodo, 2016-12-16:
I am no longer using MOOS with Morse.  I'm not sure anyone is maintaining
the pyMOOS bindings either.  I think replacing the current implementation
with something MOOSv10 compatible is the way to go.

David

severin-lemaignan, 2016-12-16:
Ok, thanks @davidhodo  for the feedback.
@msis please go ahead with a PR, then!

msis, 2017-02-09:
Hi guys!
I finally had some time to finish the migration and create the PR #757 .

PR #757 (2017-02-22): Updates middleware, tests, and docs to support new MOOS interface
This closes #752 .
closed 2017-02-22 by pierriko