morse-simulator/morse #753: Starting simulation path resolveing with morse run simulation and direct execution of the builder script

cehberlin, 2017-01-10:
Hi,
from my understanding it is possible to start a simulation "test" either by running `morse run test` after the directory "test" with a default.py has been imported or by directly executing the builder script file in such a simulation folder (actually using morseexec). However, in my builderscript I have a custom blender file for the environment that is located in "data/whatevercustom.blend" inside this simulation directory "test". If I run `morse run test` everything works as expected. If I do `./test/default.py` the blender file cannot be found: 

```
FileNotFoundError: Environment 'whatevercustom.blend' not found
Traceback (most recent call last):
  File "/tmp/test/default.py", line 13, in <module>
    env = Environment('whatevercustom.blend', fastmode = False)
  File "/usr/local/lib/python3/dist-packages/morse/builder/environment.py", line 53, in __init__
    self.append_meshes()
  File "/usr/local/lib/python3/dist-packages/morse/builder/abstractcomponent.py", line 680, in append_meshes
    filepath = self._compute_filepath(component)
  File "/usr/local/lib/python3/dist-packages/morse/builder/abstractcomponent.py", line 645, in _compute_filepath
    raise FileNotFoundError("%s '%s' not found"%(self.__class__.__name__, component))
FileNotFoundError: Environment 'whatevercustom.blend' not found

```

Is that behaviour expected? Where do I have to place the environment file correctly or how do I have to specify the path?

Minimal builder file to reproduce the issue:

```python
#! /usr/bin/env morseexec
from morse.builder import *

robot = Morsy()

env = Environment('whatevercustom.blend', fastmode = False)

---
- *MORSE version: morse 1.4-38-gb3d6-dirty
- *Blender version: 2.76.0

- *Python version: 3.5.2

dgerod, 2017-01-11:
I think you should pass the global path instead of relative one, relative path is only used by environments and other data stored in the simulation directories.

nicolaje, 2017-01-11:
Hello cehberin,

you can add the path of your environment in the MORSE_RESSOURCE_PATH environment variable so that Morse knows where to find it.

See:
http://www.openrobots.org/morse/doc/1.4/user/builder.html

cehberlin, 2017-01-12:
Ok thanks for the explanation! However, I might be a possible improvement to also look into the default subfolders (simulation environment) of a script if it is directly executed.

nicolaje, 2017-01-12:
I don't think it's a good idea to look into each subfolder of the launched script for .blend files.

Instead, you could set the path relatively to your default.py environment, e.g.:
env = Environment('data/whatevercustom.blend', fastmode = False)

If your simulation environment is in your "sites", (in $HOME/.morse/config ), for example if you created your environment with "morse create", the path to your environment will be automatically added to the MORSE_RESOURCE_PATH environment variable and your environment .blend file will load correctly.