morse-simulator/morse #755: VideoCamera problem!

alealencar, 2017-01-22:
Hi
I'm working with Blender and Morse and have my own Robot class.
I follow the www..tutorialsforblender3d.com to make a car.
If I use the video camera sensor my code fails, otherwise my code runs ok.
The error returned is:
AttributeError: 'NoneType' object has no attribute 'getConstraintId'

I don't know why, but if I use VideoCamera, the function getContstraindId returns NoneType.

My configuration is:
Ubuntu 16.04
Morse 1.4
Blender 2.76b
Python 3.5.2

Minimal builder file to reproduce the issue:
run.py

import sys
sys.path.append("../xicaum")

from AstraBuilder import *
from morse.builder import *

astra = AstraBuilder()
astra.add_service('socket')

cameraL = VideoCamera()
astra.append(cameraL)

env = Environment("../teste8/city7.blend")
env = Environment("empty")
env.properties(longitude = 0.0, latitude = 0.0, altitude = 0.0)

AstraBuilder.py

import sys
#sys.path.append("../wl")
sys.path.append("../xicaum")

#from CarClass import *
#from NetworkInterfaceCard import *
from morse.builder import *

class AstraBuilder(Robot):
	def __init__(self, name = None):
		Robot.__init__(self, "astra.blend", name)		

		#add odometry
		odometry = Odometry()
		#odometry.translate(0,0,0)
		self.append(odometry)
				
		pose = Pose()
		#pose.translate(0,1.3,0.2)
		self.append(pose)		
				
		#add gps
		gps = GPS()
		#gps.translate(0,1.3,0.2)
		self.append(gps)
				
		#add imu
		imu = IMU()
		#imu.translate(0,1.3,0.2)
		self.append(imu)
				
		#velocity
		velocity = Velocity()
		#velocity.translate(0,1.3,0.2)
		self.append(velocity)
		
		print("end builder CarAstraBuilder")

[astra.blend.zip](https://github.com/morse-simulator/morse/files/722263/astra.blend.zip)

nicolaje, 2017-01-23:
Hi Alealencar,

inside your .blend file there are 4 python scripts (Suspension.py, getRound.py, CarSetup.py, CarAstraController.py). I have had troubles using files like that: when launching MORSE, blender couldn't find these scripts. Try extracting them in your file system (eg: /home/you/scripts/... ), put their location inside your PYTHONPATH and see if the error message changes.

alealencar, 2017-01-23:
Hi Nicolaje,

I've tried what you said, but I get the same result.

 File "CarSetup.py", line 75, in Car_Constraint
    constraint_ID = vehicle_Constraint.getConstraintId()
AttributeError: 'NoneType' object has no attribute 'getConstraintId'