morse-simulator/morse #756: Create specific project in Github for Q&A

dgerod, 2017-01-25:
I think that some times we are posting here user questions and problems instead of real simulation issues. Therefore, user questions and simulation issues are mixed.

I know there is already a mailing-list for user questions and I am using it but in my opinion it is better using Github issues, in fact this is already happening. However, we should avoid to mix Q&A with simulation issues, so I propose to create another project under morse-simulation for Q&A. I found some people already doing this in Github, see: https://github.com/robotology/QA 

What do you think?

pierriko, 2017-01-25:
Hi Diego,

Labels are just great at keeping issues / reports / comments neat.

I think that another repo would add noise, having to many ways to communicate often end up behing counter beneficial.

Cheers,

dgerod, 2017-01-25:
Hi,

I understand your point of view, but what about using a defined label as "Q&A"? It will be easy to find user doubts from all issues. In addition, I think that only contributors have right to add labels at this moment, am I right?

I have a second question, do you agree that users as me ask their questions here? I would like to know where I have to send my doubts, now I am using the mailing list but I would prefer to write them here.

Thanks.

pierriko, 2017-01-25:
The Q&A label seems a very good proposition.

I have to check the permission policy on GitHub, but since the main maintainers are quite busy these days, I'd vote for a participative repository where people like you, who demonstrate interest and volunteer in helping out, could get the credentials to work on the project.

Then, on the place to ask, I don't really mind, as long as everyone feel free to communicate, anywhere is fine to me. We shouldn't give to much restriction which could scare potential newcomers away. Indeed to the extent that it doesn't generate too much noise.

pierriko, 2017-01-25:
... just to complete my comment about "noise" it's in a positive spirit to welcome everyone to join, without having to dig into a maze of information. As an open-source project, it's vital to make everyone feel quickly part of it, and this comes in different ways, documentation and communication means are the most important.

dgerod, 2017-02-01:
OK, thanks for adding the Q&A label. And I agree with your view about collaborating.

closed 2017-02-15 by dgerod