morse-simulator/morse #762: VideoCamera messes with scene background

nicolaje, 2017-03-29:
Hi guys,

using Blender 2.76, latest build of MORSE (also tested with 1.3), using a **VideoCamera** sensor makes the **background of the scene** to be a "**view**" from another Camera.. To explain, see the following GIF (builder and specific setup is described after)

[image]

On that .gif:
* when I launch the simulation we can see in the background the cube that is located in (0, 0, -7), and we see the ATRV robot as it falls down to the cube. 
* When the robot is on the cube, I press F11 to relaunch the simulation
* The "ghost background" seems to be static now, we don't see the ATRV falling
* The ATRV disapears from the "ghost background"
* I press F11 again
* The ATRV reappears after a while on the background

What I want is **not to have that "ghost background" at all**, as I have a custom skybox that gets messed up by this. I think it has to do with the "background scene" we can set in the BGE, but couldn't converge to a solution.

Minimal builder file to reproduce the issue:

```python
from morse.builder import *

robot = ATRV()
videocamera = VideoCamera()
robot.append(videocamera)

env = Environment('forssea_sim/data/forssea_sim/environments/test_empty.blend', fastmode=False)

```

Note that the "ghost background" seems to be pictured by a camera at (0, 0, 0), looking downward. Therefore, it appears only when there are objects below the Blender origin. Attached in the zipfile you will find the .blend file from the GIF.

---
- *MORSE version:* [morse 1.4-43-gcff0-dirty]
- *Blender version:* [2.76.0]
- *Python version:* [3.5.2]

[test_empty.blend.zip](https://github.com/morse-simulator/morse/files/879381/test_empty.blend.zip)

adegroote, 2017-03-30:
It looks like a change in blender internal (we use background scene to render camera since 1.1 I guess). By chance, can you try with an older version of Blender ?

nicolaje, 2017-03-30:
I couldn't try with an older version of Blender because of some Python 3.5 incompatibility, but I tested with newer versions of Blender: it starts to "work" again from Blender 2.78.

The VideoCamera seems to be rendering correctly, there seems to be no interaction with the background, **however** using ROS and trying to set the Simulator view to the VideoCamera view (pressing F9) causes the following error:

```bash
[    1.180] Showing view from camera: 'robot.camera.CameraRobot'
Python script error - object 'robot.camera.', controller 'Python':
[ERROR][MORSE] Uncaught exception, quit Blender.
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/morse/blender/calling.py", line 54, in sensor_action
    component_action(contr)
  File "/usr/local/lib/python3/dist-packages/morse/blender/calling.py", line 50, in component_action
    cmpt_object.action()
  File "/usr/local/lib/python3/dist-packages/morse/core/sensor.py", line 95, in action
    function(self)
  File "/usr/local/lib/python3/dist-packages/morse/middleware/ros/video_camera.py", line 48, in default
    image.data = bytes(image_local)
BufferError: Image buffer is not available
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/morse/blender/calling.py", line 54, in sensor_action
    component_action(contr)
  File "/usr/local/lib/python3/dist-packages/morse/blender/calling.py", line 50, in component_action
    cmpt_object.action()
  File "/usr/local/lib/python3/dist-packages/morse/core/sensor.py", line 95, in action
    function(self)
  File "/usr/local/lib/python3/dist-packages/morse/middleware/ros/video_camera.py", line 48, in default
    image.data = bytes(image_local)
BufferError: Image buffer is not available

```

Should I open a new issue for this one?

dgerod, 2017-04-02:
Hi,

I am using Blender 2.71, so I am trying to help. However, when I open your .blend file there is only a cube, see the picture.

[image]

 Could you said me what I have to do exactly to reproduce the issue?

My configuration is:
* MORSE: 1.4-40-g0c24d-dirty
* Blender: v.2.71.0
* Python: 3.4.0

nicolaje, 2017-04-02:
Hi @dgerod ,

to reproduce the issue you have to use that builder script:

```python
from morse.builder import *

robot = ATRV()
videocamera = VideoCamera()
robot.append(videocamera)

env = Environment('path/to/the/blend/file/test_empty.blend', fastmode=False)

```

with the .blend file provided (containing indeed only one cube).

I think with newer versions of Blender the Camera need some reworking, but currently this is a bit over my Blender-fu ...

pierriko, 2017-04-04:
Hello @nicolaje and @dgerod !

Since we don't have much time lately to manage PR and code review, I've just added in the Developers group on `morse-simulator` organization ;-)

Welcome aboard, feel free to push/merge (preferably after squashing), and thank you for your contributions ;-)

dgerod, 2017-04-05:
Thanks. 

Although I am quite busy with several things, I would try to do my best helping here.

dgerod, 2017-04-05:
@nicolaje, I have just tested your script and there is the problem, see below the screen-shots:

Using "MORSE edit"
[image]
[image]
Using "MORSE run"
[image]

My configuration is:
    MORSE: 1.4-40-g0c24d-dirty
    Blender: v.2.71.0
    Python: 3.4.0

nicolaje, 2017-04-05:
Thanks Pierrick!

Diego, now if you try with Blender 2.78 that artifact will be gone, but if you press F9 to see from the camera view while publishing the camera's frames over ROS, you will get the Image Buffer error above.

crey0, 2017-11-06:
Hi,
I observed the same behavior when pressing F9. I am running simulation in fixed timestep mode.

Pressing F9:
Blender <1.78: The timestamp of the next image is shifted by one simulation step.
Blender =1.79: Buffer error in the ROS callback.