morse-simulator/morse #767: Getting passive objects location

tomazela, 2017-09-30:
Hi,

I am working on a simulation where my robot grabs a passive object (cup), walks to a point and releases it. So, I would like to know if is there a way to get the position of a passive object after its falling? I've read some topics that was possible to use bpy (Python API of Blender), but I was not able to use it outside Blender python console :/

Tks in advance!

---
- *MORSE version:* morse 1.4
- *Blender version:* Blender 2.79 (sub 0)
- *Python version:* Python 2.7.12

severin-lemaignan, 2017-10-02:
Hello Lucas,

One way of achieving this is to attach a sensor (in your case, a pose sensor) to a `FakeRobot` (as passive objects can not hold directly sensors), and attach the `FakeRobot` to your object:

Something like that:

```python
chair = PassiveObject('props/objects','RollingChair')
chair.properties(Object = True)

fakerobot = FakeRobot()
chairpose = Pose()

chair.append(fakerobot)
fakerobot.append(chairpose)
fakerobot.add_default_interface('socket')
```

Let us know whether this work for you, and if it does, please close this issue.

Best,
Severin

On 30/09/17 22:08, Lucas Tomazela wrote:
> Hi,
>
> I am working on a simulation where my robot grabs a passive object (cup), walks to a point and releases it. So, I would like to know if is there a way to get the position of a passive object [...]

________________________________
[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, Plymouth University accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. Plymouth University does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

tomazela, 2017-10-04:
Hi Severin!

This workaround has worked :) I have only a remain question: looking into Pose actuator on Morse Doc, I can access pose data through get_local_data() method. In my python script, I have realized that calling this method returns a Morse Object, but is there an easy way to retrieve position coordinates (x, y, z) from it?

Regards,
Lucas

tomazela, 2017-10-08:
Hi Severin,

This workaround has worked :) 

Thanks!
Lucas

closed 2017-10-08 by tomazela