morse-simulator/morse #768: Simulating wind force

tomazela, 2017-10-08:
Hi all,

I am working in a simulation where I have a quadcopter which grabs an object, goes to a waypoint and releases it. In order to turning this simulation closer to real word, I am trying to simulate how wind can influence object falling (so wind force would be only applied on object after quad releases it), but I have not been succeed :(
Is there someone who can help me?

Thanks,
Lucas

--
**Server**
```python
from morse.builder import *

# Dynamic quadrotor
quad = Quadrotor()
quad.translate(x=0, y=0, z=0.5)
quad.add_default_interface('socket')

# Gripper
gripper = Gripper()
gripper.translate(z=-0.2)
gripper.properties(Angle=360.0, Distance=2.0)
quad.append(gripper)

# Waypoint
waypoint = RotorcraftWaypoint()
quad.append(waypoint)

# Wind
wind = ExternalForce()

# Object
cupObject = PassiveObject('props/kitchen_objects', 'Cup_Blue')
cupObject.translate(x=0, y=0, z=0)
cupObject.properties(Object=True)
cupObject.setgraspable()
cupRobot = FakeRobot()
cupObject.append(cupRobot)
poseCup = Pose()
cupRobot.append(poseCup)
cupRobot.append(wind)
cupRobot.add_default_interface('socket')

environment = Environment('land-1/trees')

```

**Client:**
```python
from pymorse import Morse

with Morse() as morse:
    # Grabbing object
    morse.quad.gripper.grab()

    # Moving quad to another point
    morse.quad.waypoint.setdest(1.0, 1.0, 4.0, 0.0, 0.1)

    # Setting external force on object
    morse.cupRobot.wind.applyForce([100.0, 0.0, 0.0])

    # Releasing object
    morse.quad.gripper.release()
```
Obs: I have modified externalForce.py, in order to be able to modify force data. So, 'applyForce' method is like:
```python
    @service
    def applyForce(self, force):
        robot = self.robot_parent.bge_object

        # Setting new forces value
        add_data('force', force, "vec3<float>", "force along x, y, z")	
        
        # Applying force
        robot.applyForce(self.local_data['force'], False)
```
---
- *MORSE version:* morse 1.4
- *Blender version:* Blender 2.79 (sub 0)
- *Python version:* Python 2.7.12

adegroote, 2017-10-20:
as far as I remind, FakeRobot do not have physics, so you cannot apply "force" on it. You need to create an object at least "dynamic" (in Blender terminology).

nicolaje, 2017-10-26:
You could also create your own actuator that makes your PassiveObject move according to the force you set